micropython/micropython #3876: SDCard concurrent use of SPI

mindforger, 2018-06-18:
i am trying to wrap my head around https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py but i have a hard time to figure out which calls al interrupt driven and which are sequential

i want to understand, if i were to drop a script on the SDCard (a picoweb server in my case) and i want to use the SPI to drive some APA102, i need to use another CS pin for them.

And I have all wired up already but i am not sure how i may interfere with SDCard SPI or if there is a methode inside the SDCard.py available to ask if i can use the SPI for a brief moment.

I need to shovel 1200Bytes + some dummy bytes without interruption through the SPI but i need to load them from the SD first and don't want to send bullshit to the SDCard just because their CS is still active.

dpgeorge, 2018-06-19:
> which calls are interrupt driven and which are sequential

It depends on the port/board which you are using.  All calls from the VFS sub-system to this sdcard driver are *user instigated*.  Usually that means that it's sequential, that the driver is only called if you explicitly make a Python call to open/read/write/close/etc a file on the SD card.

But in some cases this is not true: on esp8266 for example if you are running WebREPL in the background and you access a file on the SD card via WebREPL then it may interrupt your code.  This is still user instigated, but can be asynchronous to your Python script.

BTW, this kind of question is better suited to ask on the forum: https://forum.micropython.org

mindforger, 2018-06-19:
>But in some cases this is not true: on esp8266 for example if you are running WebREPL in the background and you access a file on the SD card via WebREPL then it may interrupt your code. This is still user instigated, but can be asynchronous to your Python script.

yes i am doing practically this exactly (just with an SP8285 but only the internal flash is the difference), so as long as i don't upload files it should be okay then, thank you

>BTW, this kind of question is better suited to ask on the forum: https://forum.micropython.org
--

i know, but when you got compromised somewhere along your online lifetime you start to prefere to have less accounts and less passwords to manage ... i now prefere SSO with offline token, i wish you never have to deal with compromised personal data

closed 2018-06-19 by mindforger