micropython/micropython #3878: Ethernet with STM NUCLEO-F429ZI

Abdelrahman-Elnaggar, 2018-06-19:
Hello,
I would like to ask if the Ethernet connection is supported with STM NUCLEO-F429ZI ? 
If it is supported is there is any sample code or anything that could help me to start using it ?
i would really appreciate any help.
thanks in advance

dpgeorge, 2018-06-19:
There is preliminary support for ETH in PR #3808.  It gets ETH working with two F7 boards.  It should be pretty straightforward to take that PR and add support for the NUCLEO-F429ZI, it just requires updating the board config files.

Abdelrahman-Elnaggar, 2018-06-20:
thanks dpgeorge so much for your help ,as i will start updating the board config files ,
but do you know any sample code or guide on how to write a simple code  to test the Ethernet after i update the files ?? 
i would really appreciate the help.

dpgeorge, 2018-06-20:
> but do you know any sample code or guide on how to write a simple code to test the Ethernet after i update the files ?

The "usual" way of using networking in MicroPython should just work with the Ethernet, see eg http://docs.micropython.org/en/latest/pyboard/library/network.html#module-network

For example:
```python
import network
eth = network.Ethernet()
eth.active(1)
eth.ifconfig('dhcp')
import socket
socket.getaddrinfo('micropython.org', 80) # should return the correct IP address of this site
```

Abdelrahman-Elnaggar, 2018-06-21:
> The "usual" way of using networking in MicroPython should just work with the Ethernet, see eg http://docs.micropython.org/en/latest/pyboard/library/network.html#module-network
> 
> For example:
> 
> import network
> eth = network.Ethernet()
> eth.active(1)
> eth.ifconfig('dhcp')
> import socket
> socket.getaddrinfo('micropython.org', 80) # should return the correct IP address of this site

But when i write eth = network.Ethernet() in the serial prompt i get this error : 
AttributeError: 'module' object has no attribute 'Ethernet'

dpgeorge, 2018-06-22:
> But when i write eth = network.Ethernet() in the serial prompt i get this error ...

You need to make sure you have pulled the PR #3808 and in the config file for your new board put `#define MICROPY_HW_ENABLE_ETH_RMII  (1)`

dpgeorge, 2018-06-23:
> but i managed to solve these errors by changing the address
> and the second error SCB_CleanInvalidateDcache(); i found out that this function is specific for f7 boards so i commented this line,

That sounds correct.

> in the serial prompt i get the same error as before
> AttributeError: 'module' object has no attribute 'Ethernet'

I'm not sure what is wrong here.  If you can please post your code (all the additional files you added and any changes you made to existing files) then it will be easier to debug it.

Abdelrahman-Elnaggar, 2018-06-25:
thank you @dpgeorge  for your help as i managed to solve the problem by writing these two commands to get the firmware:
make -C mpy-cross
make -C ports/stm32 MICROPY_HW_ENABLE_ETH_RMII=1 BOARD=NUCLEO_F429ZI

and upload it normally to the board

closed 2018-07-03 by Abdelrahman-Elnaggar

YO3HCM, 2022-03-29:
Hi everyone, 
I just put micropython on my STM32F767ZI nucleo board. it works fine.
I tried ethernet, and I get a timeout on the third line... 
`import network
eth = network.Ethernet()
eth.active(1)`

Do you have any advice?

Cheers