micropython/micropython #3884: STM32F4_DISCO Conflict Flash memory and extra sector (11) storage

rolandvs, 2018-06-20:
In `port/stm32` the `flashbdev.c` has a provision to add another sector of 128kByte to the file system.
Unfortunately this space is already "taken" by the memory reserved for the micropython firmware (see `stm32f405.ld`. Most user probably will not have noticed or do not use this.

dpgeorge, 2018-06-23:
> Unfortunately this space is already "taken" by the memory reserved for the micropython firmware

Yes that's indeed true.  Well, the intention is that this is configurable by the user for their particular board, and they will need to be aware that if the last 128k is used for the filesystem then the firmware must fit before that.  My suggestion to "fix" this would be to simply add a comment in flashbdev.c to document the above fact.

rolandvs, 2018-06-24:
@dpgeorge If possible would it be possible to add a flag other than `#if(0)` so this feature can be reached outside the driver and without "changing" the mpy code... There are some more instances in the driver for the second segment use. I have made some different versions (see #3752) that extended beyond the segment #2 and leave the original code as it is, and enabling different memory options for the /flash drive by setting `#define MICROPY_HAS_FLASH_EXTRA` as the deviation from the standard and copied/moved the global-ish linker file into the boards directory (IMHO the place where it should be :-).

dpgeorge, 2018-06-25:
> Would it be possible to add a flag other than #if(0) so this feature can be reached outside the driver and without "changing" the mpy code

I still prefer the more comprehensive and flexible solution that's discussed in #3752.

rolandvs, 2018-06-25:
OK, I will put it on my list to take a go at it...

(meanwhile could you take a slice of your precious time to glance at #3883?)

dpgeorge, 2022-03-30:
Flash configuration was made more flexible in 84969194a18ec015d2508db14be66457f123f687

closed 2022-03-30 by dpgeorge