micropython/micropython #3886: python3 compat: assignment and evaluation order

pmp-p, 2018-06-21:
hi, expecting :

```
Python 3.7.0rc1 (default, Jun 18 2018, 19:28:58) 
[GCC 4.8.5] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> m='os'
>>> m = sys.modules[m] = __import__(m)
>>> m
<module 'os' from '/usr/local/lib/python3.7/os.py'>

```

got:
```
MicroPython v1.9.4-114-g309fe39-dirty on 2018-06-20; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import sys
>>> m='os'
>>> m = sys.modules[m] = __import__(m)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __hash__: 'module'
```

dpgeorge, 2018-06-21:
I don't think this is a problem with assignment ordering.  The docs https://docs.python.org/3/reference/simple_stmts.html#assignment-statements state that assignment happens left-to-right, so:
```python
a = b = expr
```
is equivalent to:
```python
temp = expr
a = expr
b = expr
```
(this is sensible because you read it left-to-right, and it matches `i, a[i] = 1, 2` which sets `i=1` first, then `a[1]=2`).

For example:
```python
m = 1
d = {}
m = d[m] = m + 1
print(d)
```
gives `{2: 2}` in both CPython and uPy.

Note that uPy doesn't support hashing of modules (as per the error reported above), so that's why it differs to CPython, but I assume you didn't really want to have the module itself as the key in the dict.

peterhinch, 2018-06-21:
> temp = expr
a = expr
b = expr

I think you intended:
```python
temp = expr
a = temp
b = temp
```

pmp-p, 2018-06-21:
indeed it's because of module not hashable, does not matter much then ...

closed 2018-06-21 by pmp-p