micropython/micropython #3891: Caught exception socket.error : 'ussl' object has no attribute 'makefile'

Sunrise17, 2018-06-24:
For my following socket connection below, i am receiving  'ussl' object has no attribute 'makefile' error in due to line `f = sock.makefile('rwb', buffering=0)` . Version is MicroPython v1.9.4-199-g6fc84a74 on 2018-06-24; ESP32 module with ESP32.

        s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)
        addr = usocket.getaddrinfo(HOST, PORT)[0][-1]
            s.connect(addr)
            sock = ussl.wrap_socket(s)
            lines = [
                'GET %s HTTP/1.1' % targetName,
                'Host: %s:%s' % (HOST, PORT),
                'Connection: close',
                ]

            f = sock.makefile('rwb', buffering=0)
            sendObject = '\r\n'.join(lines)+'\r\n\r\n'
            bytesSendObject = sendObject.encode()
            
            f.write(bytesSendObject)
            recvData = f.read(4096)
            f.close()
            sock.close()

peterhinch, 2018-06-24:
The `usocket` module, by virtue of being a 'micro' implementation, lacks the `socket.makefile` method.

Sunrise17, 2018-06-24:
Ok @peterhinch , thanks for the information.

closed 2018-06-24 by Sunrise17