micropython/micropython #3894: ESP32 interrupt latency is long and irregular

nickovs, 2018-06-24:
Unlike on other micropython ports, on the ESP32 the time between a hardware interrupts occurring and Python handlers being called is irregular and potentially very long. This can be shown using the following test code, a decent oscilloscope and a push button:
```
p_in = Pin(21, Pin.IN)
p_out = Pin(17, Pin.OUT)
def handler(p):
    p_out.value(p_in.value())
p_in.irq(handler=handler, trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING)
```
On the ESP8266 the time between edges on the input and output pins is about 140us and very consistent. On the ESP32 the time between edges is varies between a few tens of microseconds and 10 milliseconds.

I have recently been adding code to make use of the higher resolution timers in these two devices and the same issues occur with timer interrupts; on the ESP8266 timer events are delivered in a timely manner, with latency around 150 microseconds, while on the ESP32 the latency can be up to ten milliseconds.

I've not delved into the underlying code but based on the observed timing my guess is that the handlers on the ESP32 are not being called asynchronously (as happens on other platforms) but instead are being _scheduled_ to happen later and then having to wait for some 100Hz timer to notice that there is something to do. This renders the ESP32 basically useless for any function that requires accurate timing of signals. There are plenty of cases where low and consistent interrupt latency is important even when overall performance is not needed; an example would be building an AC dimmer using a zero-cross detector and a triac.

It would be good to find a way to have interrupt handlers on the ESP32 have consistent and low latency.

dpgeorge, 2018-06-25:
> This can be shown using the following test code, a decent oscilloscope and a push button ...

I tried out this test code and did see a lot of 5-10ms delays from change of pin 21 to change of pin 17.  But (in my case at least) it looks like this is a debouncing problem.  The button was not debounced and the `handler()` was being scheduled many times per "press".  It looks like the last call to `handler()` was the one dictating the output on pin 17, and this was introducing the delay.

I modified the C code in `esp32/machine_pin.c:machine_pin_isr_handler` to explicitly (in C) copy the value of pin 21 to pin 18 (eg `gpio_set_level(18, gpio_get_level(21))`) as well as call the Python handler, and this change of pin 18 seems to have much lower latency.

> I have recently been adding code to make use of the higher resolution timers in these two devices and the same issues occur with timer interrupts;

OK, so that probably debunks the debouncing theory above :)

> I've not delved into the underlying code but based on the observed timing my guess is that the handlers on the ESP32 are not being called asynchronously (as happens on other platforms) but instead are being scheduled to happen later

Yes this is true, the call `mp_sched_schedul(...)` schedules the Python handler to run "later", which will be as soon as the VM notices there is something pending.  But this is how ESP8266 does it as well.  So that points at the ESP32's RTOS (namely FreeRTOS) and for some reason it introduces delays not seen on ESP8266.

robert-hh, 2018-06-25:
See also the discussion here https://github.com/micropython/micropython/issues/2972
and here for a similar set-up: https://forum.pycom.io/topic/936/pin-interrupt-latency

peterhinch, 2018-06-25:
@robert-hh 10ms is an order of magnitude worse than the figures we were seeing, and you did measurements on a LoPy. I wonder if there has there been some change (perhaps in the vendor code) to cause this?

robert-hh, 2018-06-25:
@peterhinch Murphy: Never say it cannot be worse. 
Maybe I did to set the time range wide enough.

nickovs, 2018-06-25:
OK, I did a little more investigation and it seems that the problem is in the way that the ESP32 handles scheduling when when Python code is sleeping. Specifically, at the heart of the implementation of `sleep_ms()` is a call to `vTaskDelay(1)`, which _always_ sleeps for one 100Hz _tick_, before performing another call to `MICROPY_EVENT_POLL_HOOK`. If I replace my call to `time.sleep()` with a microsecond-timed spin loop that calls `sleep_us()` in the middle (`sleep_us()` itself being a spin loop that checks `mp_handle_pending()`) then the interrupt latency drops to an reasonable level with fairly reliable performance and my high frequency timer code works just fine on multi-kilohertz timers.

While using a spin loop works it might not be the most power-efficient way to do things. What we probably need here is some sort of RTOS call to wait for a semaphore with a timeout, and set the semaphore whenever something is added to the micropython pending list. The FreeRTOS _Task Notifications_ look like them might be able to do what we need (perhaps [TaskNotifyGiveFromISR](https://www.freertos.org/vTaskNotifyGiveFromISR.html)) and there also appears to be a whole semaphore system too; what I don't know is if the implementations end up waiting for ticks before switching tasks. Perhaps someone with better knowledge of FreeRTOS can suggest the best way to do this. Alternatively we could just implement all sleep functions with a spin-loop and not worry about it.

dpgeorge, 2018-06-26:
> Specifically, at the heart of the implementation of sleep_ms() is a call to vTaskDelay(1), which always sleeps for one 100Hz tick, before performing another call to MICROPY_EVENT_POLL_HOOK

Ok, that makes sense then, why there is about 10ms to service some of the pending event handlers.

> While using a spin loop works it might not be the most power-efficient way to do things.

I checked, and a spin loop takes about 19mA more than a vTaskDelay (37mA increases to 56mA, running at 240MHz).  So it would be best to make it power efficient.

> What we probably need here is some sort of RTOS call to wait for a semaphore with a timeout, and set the semaphore whenever something is added to the micropython pending list

That sounds like a good idea.

> what I don't know is if the implementations end up waiting for ticks before switching tasks

I don't know either, but I'd guess that if the system was idle then it would schedule any tasks that become runable straightaway.  So if `time.sleep(n)` is sitting there waiting for a semaphore (or task notify) and doing nothing else, and an IRQ comes along and signals the semaphore, then when the IRQ finishes it should wake the waiting task immediately.

nickovs, 2018-06-26:
OK. A 50+% increase in power consumption is worth avoiding if we can. 

I could hack something together to set a semaphore when tasks are added to the pending list and wait for the semaphore when sleeping but it might be good to define a proper set of hooks for this that can be used in a cross-platform way. Given that this is going to hook into the core of the VM it might be better if you define that rather than me guessing, although I can take a stab some time if you want. If I do it then your input on where I might want to wait on a semaphore would be helpful. It looks like the implementations of `select()` poll in a loop rather than wait in the RTOS so maybe `sleep_ms()` is the only place that needs it. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nickovs, 2018-06-28:
Thanks for the input. I had found most of that but had totally missed the need to wait on the semaphore in `mp_hal_stdin_rx_chr()`.

I was thinking we might want some sort of hook macro in the body of `mp_sched_schedule()` that can call a platform-specific function to set the semaphore (since the scheduler is not in the HAL). Then we can replace the waiting functions in `mp_hal_stdin_rx_chr()` and `mp_hal_delay_ms()` (which currently both call `vTaskDelay()`) with calls that will wait on the semaphore.

I will have a go at building something for the ESP32 and then we can look at generalising from there. It might need to wait until next week though.

dpgeorge, 2018-07-04:
Now fixed by 14ab81e87accedfb9ed231b206dd21f3a0143404

closed 2018-07-04 by dpgeorge