micropython/micropython #3897: ESP8266 Crash if ctrl-C issued in file I/O with two files open

peterhinch, 2018-06-25:
This arose [in this forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=4928&p=28335#p28335). Tested on Adafruit Feather Huzzah.

To replicate, on an ESP8266 which is connected in station mode create a test file:
```python
with open('syslog', 'w') as f3:
    for _ in range(10):
        f3.write('testdata\n')
```
Reboot, run this script and interrupt with ctrl-C:
```python
while True:
    with open('syslog', 'r') as f2:
        with open('syslog.1', 'w') as output:
            for line in f2:
                output.write(line)
```
It seems only to occur with two nested context managers. Response is
```
Fatal exception 0(IllegalInstructionCause):
epc1=0x4027742c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)
```

dpgeorge, 2018-06-25:
Thanks for reporting.  Is it possible to get the crash by explicitly raising an exception (eg `raise KeyboardInterrupt`) after `output.write(line)`, or does it need to come from a ctrl-C?

dpgeorge, 2018-06-26:
Ok I can confirm the bug.  The issue is that the UART IRQ is executed while the flash is being read/written, and so the iROM cache is disabled, and then the call to `mp_keyboard_interrupt` crashes the device because this latter function is in iROM (address epc1=0x4027742c above).

Another way to trigger the same bug is:
```python
import esp
b = bytearray(8192)
for i in range(1000):
    esp.flash_read(i, b)
```
Then press ctrl-C while the loop executes.

One fix is to disable UART IRQs while the flash is reading/writing.  Another fix would be to put `mp_keyboard_interrupt()` in iRAM.

dpgeorge, 2018-06-28:
Fixed by d800ed187756c734458d269b1bf0dbfe554ad4f1

closed 2018-06-28 by dpgeorge

peterhinch, 2018-07-01:
@dpgeorge The user implemented your fix. He now reports that this crashes spontaneously after a period without issuing ctrl-C. I'm loath to try to replicate this out of sympathy for the flash :)
```python
while True:
    with open('syslog', 'r') as f2:
        with open('syslog.1', 'w') as output:
            for line in f2:
                output.write(line)
```
Reference: [Forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=4928&start=9).

dpgeorge, 2018-07-03:
> He now reports that this crashes spontaneously after a period without issuing ctrl-C.

This should be now fixed by b6e5f82ba524b953d5b93bc9713bb0a4ef55489b