micropython/micropython #3900: micropython.elf failing to build for ports/teensy on Windows

dr5801, 2018-06-26:
- Windows version: 10
- Cygwin version: 10.0
- Arm toolchain version: 7 2017-q4-major

### Description:
I have a Teensy 3.5 board that I'm trying to load micropython onto from Windows.  Following the steps from the [building section of the wiki](https://github.com/micropython/micropython/wiki/Board-Teensy-3.1-3.5-3.6#building), I have not been able to successfully build the micropython.elf file.

### What I've run:

- [In Cygwin](https://gist.github.com/dr5801/5f84c15b92d7a714f4486cf9e13871da)
- [In CMD](https://gist.github.com/dr5801/7f79e4f997e29acd48a07a63c8e769cb)

<details>
<summary> Cygwin </summary>

```
darife@ACC-BKZ2FH2 /cygdrive/d/git/micropython/micropython/ports/teensy
$ which arm-none-eabi-gcc
/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc

darife@ACC-BKZ2FH2 /cygdrive/d/git/micropython/micropython/ports/teensy
$ which mingw32-make
/cygdrive/d/MinGW/bin/mingw32-make

darife@ACC-BKZ2FH2 /cygdrive/d/git/micropython/micropython/ports/teensy
$ which make
/cygdrive/d/MinGW/msys/1.0/bin/make

darife@ACC-BKZ2FH2 /cygdrive/d/git/micropython/micropython/ports/teensy
$ echo $PATH
/usr/local/bin:/usr/bin:/cygdrive/c/Program Files/Microsoft MPI/Bin:/cygdrive/c/Perl64/site/bin:/cygdrive/c/Perl64/bin:/cygdrive/c/Program Files/SlickEdit Pro 21.0.3/win:/cygdrive/d/MinGW/msys/1.0/bin:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files (x86)/IBM/Client Access/Emulator:/cygdrive/c/Program Files (x86)/IBM/Client Access/Shared:/cygdrive/c/Program Files (x86)/IBM/Client Access:/cygdrive/c/Program Files (x86)/Common Files/EMC:/cygdrive/c/Program Files (x86)/IBM/Lotus/Notes:/cygdrive/c/Program Files/PuTTY:/cygdrive/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit:/cygdrive/c/Program Files/TortoiseSVN/bin:/cygdrive/c/Program Files/MySQL/MySQL Utilities 1.6:/cygdrive/c/Program Files (x86)/WebEx/Productivity Tools:/cygdrive/c/Program Files/dotnet:/cygdrive/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin:/cygdrive/d/MinGW/bin:/cygdrive/c/Users/darife/AppData/Local/Programs/Python/Python36/Scripts:/cygdrive/c/Users/darife/AppData/Local/Programs/Python/Python36:/cygdrive/c/Users/darife/AppData/Local/Microsoft/WindowsApps:/cygdrive/d/Programs/Arduino_Install/arduino-1.8.5-windows/arduino-1.8.5

darife@ACC-BKZ2FH2 /cygdrive/d/git/micropython/micropython/ports/teensy
$ mingw32-make BOARD=TEENSY_3.5
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Using toolchain from PATH
mkdir -p build/genhdr
Create build/genhdr/pins.h
GEN build/frozen.c
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/
mkdir -p build/build/
mkdir -p build/core/
mkdir -p build/extmod/
mkdir -p build/lib/embed/
mkdir -p build/lib/libc/
mkdir -p build/lib/mp-readline/
mkdir -p build/lib/utils/
mkdir -p build/ports/stm32/
mkdir -p build/py/
CC ../../py/mpstate.c
CC ../../py/nlr.c
CC ../../py/nlrx86.c
CC ../../py/nlrx64.c
CC ../../py/nlrthumb.c
CC ../../py/nlrxtensa.c
CC ../../py/nlrsetjmp.c
CC ../../py/malloc.c
CC ../../py/gc.c
CC ../../py/pystack.c
CC ../../py/qstr.c
CC ../../py/vstr.c
CC ../../py/mpprint.c
CC ../../py/unicode.c
CC ../../py/mpz.c
[... 187 lines cut ...]
Using toolchain from PATH
GEN build/frozen.c
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR not updated
mkdir -p build/
mkdir -p build/build/
mkdir -p build/core/
mkdir -p build/extmod/
mkdir -p build/lib/embed/
mkdir -p build/lib/libc/
mkdir -p build/lib/mp-readline/
mkdir -p build/lib/utils/
mkdir -p build/ports/stm32/
mkdir -p build/py/
CC build/frozen.c
LINK build/micropython.elf
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `arm-none-eabi-gcc -nostdlib -T mk20dx256.ld -msoft-float -mfloat-abi=soft -Wl,--gc-sections -o "build/micropython.elf" -Wl,-Map,build/micropython.map build/py/mpstate.o build/py/nlr.o build/py/nlrx86.o build/py/nlrx64.o build/py/nlrthumb.o build/py/nlrxtensa.o build/py/nlrsetjmp.o build/py/malloc.o build/py/gc.o build/py/pystack.o build/py/qstr.o build/py/vstr.o build/py/mpprint.o build/py/unicode.o build/py/mpz.o build/py/reader.o build/py/lexer.o build/py/parse.o build/py/scope.o build/py/compile.o build/py/emitcommon.o build/py/emitbc.o build/py/asmbase.o build/py/asmx64.o build/py/emitnx64.o build/py/asmx86.o build/py/emitnx86.o build/py/asmthumb.o build/py/emitnthumb.o build/py/emitinlinethumb.o build/py/asmarm.o build/py/emitnarm.o build/py/asmxtensa.o build/py/emitnxtensa.o build/py/emitinlinextensa.o build/py/formatfloat.o build/py/parsenumbase.o build/py/parsenum.o build/py/emitglue.o build/py/persistentcode.o build/py/runtime.o build/py/runtime_utils.o build/py/scheduler.o build/py/nativeglue.o build/py/stackctrl.o build/py/argcheck.o build/py/warning.o build/py/map.o build/py/obj.o build/py/objarray.o build/py/objattrtuple.o build/py/objbool.o build/py/objboundmeth.o build/py/objcell.o build/py/objclosure.o build/py/objcomplex.o build/py/objdeque.o build/py/objdict.o build/py/objenumerate.o build/py/objexcept.o build/py/objfilter.o build/py/objfloat.o build/py/objfun.o build/py/objgenerator.o build/py/objgetitemiter.o build/py/objint.o build/py/objint_longlong.o build/py/objint_mpz.o build/py/objlist.o build/py/objmap.o build/py/objmodule.o build/py/objobject.o build/py/objpolyiter.o build/py/objproperty.o build/py/objnone.o build/py/objnamedtuple.o build/py/objrange.o build/py/objreversed.o build/py/objset.o build/py/objsingleton.o build/py/objslice.o build/py/objstr.o build/py/objstrunicode.o build/py/objstringio.o build/py/objtuple.o build/py/objtype.o build/py/objzip.o build/py/opmethods.o build/py/sequence.o build/py/stream.o build/py/binary.o build/py/builtinimport.o build/py/builtinevex.o build/py/builtinhelp.o build/py/modarray.o build/py/modbuiltins.o build/py/modcollections.o build/py/modgc.o build/py/modio.o build/py/modmath.o build/py/modcmath.o build/py/modmicropython.o build/py/modstruct.o build/py/modsys.o build/py/moduerrno.o build/py/modthread.o build/py/vm.o build/py/bc.o build/py/showbc.o build/py/repl.o build/py/smallint.o build/py/frozenmod.o build/extmod/moductypes.o build/extmod/modujson.o build/extmod/modure.o build/extmod/moduzlib.o build/extmod/moduheapq.o build/extmod/modutimeq.o build/extmod/moduhashlib.o build/extmod/modubinascii.o build/extmod/virtpin.o build/extmod/machine_mem.o build/extmod/machine_pinbase.o build/extmod/machine_signal.o build/extmod/machine_pulse.o build/extmod/machine_i2c.o build/extmod/machine_spi.o build/extmod/modussl_axtls.o build/extmod/modussl_mbedtls.o build/extmod/modurandom.o build/extmod/moduselect.o build/extmod/modwebsocket.o build/extmod/modwebrepl.o build/extmod/modframebuf.o build/extmod/vfs.o build/extmod/vfs_reader.o build/extmod/vfs_posix.o build/extmod/vfs_posix_file.o build/extmod/vfs_fat.o build/extmod/vfs_fat_diskio.o build/extmod/vfs_fat_file.o build/extmod/utime_mphal.o build/extmod/uos_dupterm.o build/lib/embed/abort_.o build/lib/utils/printf.o build/hal_ftm.o build/hal_gpio.o build/help.o build/main.o build/lcd.o build/led.o build/modpyb.o build/pin_defs_teensy.o build/reg.o build/teensy_hal.o build/timer.o build/uart.o build/usb.o build/lexerfrozen.o build/build/frozen.o build/ports/stm32/gccollect.o build/ports/stm32/irq.o build/ports/stm32/pin.o build/ports/stm32/pin_named_pins.o build/ports/stm32/gchelper.o build/core/mk20dx128.o build/core/pins_teensy.o build/core/analog.o build/core/usb_desc.o build/core/usb_dev.o build/core/usb_mem.o build/core/usb_serial.o build/core/yield.o build/lib/libc/string0.o build/lib/mp-readline/readline.o build/lib/utils/pyexec.o build/lib/utils/sys_stdio_mphal.o build/pins_gen.o -L c:/ ./ (x86)/ ./ ./ embedded/ 2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/ -lm -L c:/ ./ (x86)/ ./ ./ embedded/ 2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/ -lc -L c:/ ./ (x86)/ ./ ./ embedded/ 2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/thumb/v7e-m/ -lgcc'
Makefile:191: recipe for target 'build/micropython.elf' failed
mingw32-make: *** [build/micropython.elf] Error 258
```

</details>

dr5801, 2018-06-26:
Found the issue why it wasn't building based on the suggestion by @dhylands : 

> Hi Drew,
> 
> To be honest, I've never tried building the teensy stuff under windows. Looking at your gist, I think that the problem is probably caused by the fact that the path that the toolchain is installed in contains spaces and other characters (like the opening/closing parenthesis).
> 
> The first thing I'd try is to install the toolchain into a directory which contains no spaces (perhaps something like c:\arm-toolchain. It may be as simple as copying all of the files/folders from the 2017-q4-major directory tree. I'm positive that the toolchain doesn't have any registry settings or anything like that.
> 
> Dave Hylands

micropython.elf was failing to build because I had spaces in in my path to the arm toolchain:
```
$ which arm-none-eabi-gcc
/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc
```

Thank you

closed 2018-06-26 by dr5801

HaidStefan, 2018-07-12:
Hi dr5801,

You have deleted your gists. Is it possible that you share your command for CMD and Cygwin here?

many thanks
cheers

dr5801, 2018-07-12:
Hi @HaidStefan,

I accidentally deleted the gists but I have requested that github recover them for me.  Sorry for the inconvenience.

The commands still exist in the 'Cygwin' and 'CMD' drop-downs for me.  Did you try looking at them?

Thank you

dr5801, 2018-07-12:
@HaidStefan 

The gists have been restored.

Thank you

HaidStefan, 2018-07-12:
Okay many thanks. Now it was possible for me to have a look on that gists. So you were able to build the micropython on windows 10 with cygwin. Do you also get it to work under CMD.

dr5801, 2018-07-12:
I was able to build but uploading to the teensy was not successful.  The only Teensy board that is supported when I tried was 3.1 (I have a 3.5).  I have only tried Cygwin, I did not try CMD as I figured the build would want a bash environment.

HaidStefan, 2018-07-12:
Okay many thanks. So i will try cygwin. I have already tried it under msys2 but that does not work.

dhalbert, 2018-07-12:
Not sure if this will help, but: a couple of days ago I had to build `mpy-cross` under Windows: I needed a totally statically-linked version for distribution. I succeeded by doing this (adapted from my terse notes):

* Download `mingw-get-setup.exe` from https://osdn.net/projects/mingw/
* Install **mingw**, choosing basic install (gcc) and also msys installation. Find `msys.bat` in `C:\MinGW\msys\1.0`, and make a shortcut to it. That will give you a shell with Unix-style commands. 
* Then just use `make` in the usual way.

I had a problem with missing `errno.h` constants that was not hard to fix. That may be an mpy-cross-specific issue, and not an issue for the Teensy build.

I have tried older mingw32's from sourceforge and also msys2. Executables produced by msys2 require an msys2-specific .dll so they are not completely static, which is a problem for me but maybe not for you.