micropython/micropython #3904: ESP32 , SPIRAM

FFtust, 2018-06-29:
HI,  
  I enabled SPIRAM in esp32, and then I created a python thread using _thread module, but the system crashed,  
here is the log:  

        [0;32mI (1264) spiram: SPI SRAM memory test OK[0m
        [0;32mI (1265) heap_init: Initializing. RAM available for dynamic allocation:[0m
        [0;32mI (1265) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
        [0;32mI (1271) heap_init: At 3FFC5AC8 len 0001A538 (105 KiB): DRAM[0m
        [0;32mI (1278) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
		[0;32mI (1284) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m	
		[0;32mI (1290) heap_init: At 40095524 len 0000AADC (42 KiB): IRAM[0m
		[0;32mI (1297) cpu_start: Pro cpu start user code[0m
		[0;32mI (1302) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
		[0;32mI (1310) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
		[0;32mI (213) cpu_start: Starting scheduler on PRO CPU.[0m
		mp_task_heap_size 2883540
		execute default
		../../../esp-idf/components/freertos/tasks.c:686 (xTaskCreateStaticPinnedToCore)- assert failed!
		abort() was called at PC 0x4008ed67 on core 0
		
		Backtrace: 0x40090d8b:0x3ffb6f40 0x40090dbc:0x3ffb6f60 0x4008ed67:0x3ffb6f80 0x400f46c6:0x3ffb6fc0 0x400f4725:0x3ffb7000 0x400e9015:0x3ffb7020 0x400e022e:0x3ffb7050 0x400dc369:0x3ffb7080 0x400dc3d1:0x3ffb70a0 0x400e9f1a:0x3ffb70c0 0x400e02b8:0x3ffb7160 0x400dc369:0x3ffb71d0 0x400e9e87:0x3ffb71f0 0x400e02b8:0x3ffb7290 0x400dc369:0x3ffb72f0 0x400dc396:0x3ffb7310 0x400fad52:0x3ffb7330 0x400fafd9:0x3ffb73d0 0x400f1203:0x3ffb7400
		
		Rebooting...
		ets Jun  8 2016 00:22:57
		
		rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)

according to the error logs, I modified mpthreadport.c like this:

		void mp_thread_create_ex(void *(*entry)(void*), void *arg, size_t *stack_size, int priority, char *name) {
        ...
		#if CONFIG_SPIRAM_SUPPORT
    	StaticTask_t *tcb = malloc(sizeof(StaticTask_t));
		#else /* CONFIG_SPIRAM_SUPPORT */
    	StaticTask_t *tcb = m_new(StaticTask_t, 1);
		#endif /* CONFIG_SPIRAM_SUPPORT */
		}

		void vPortCleanUpTCB(void *tcb) {
		...
		#if CONFIG_SPIRAM_SUPPORT
            free((StaticTask_t*)th->tcb);
		#else /* CONFIG_SPIRAM_SUPPORT */
            m_del(StaticTask_t, th->tcb, 1);
		#endif /* CONFIG_SPIRAM_SUPPORT */
		...
		}

then I did create a thread and ran well, but error occured when I operate files system in python thread,  
here is my test script:
	
		import time
		import uos
		import _thread
		
		def filesystem_test1():
		    print("filesystem_test")
		    cwd = uos.getcwd()
		    dirs = uos.listdir()
		    uos.chdir('/')
		    uos.chdir(cwd)
		    for d in dirs:
		        print("dir:", d)
		    uos.chdir('/lib')
		    if uos.stat('request_common_neurons_command.csv'):
		        print("request_common_neurons_command exsit")
		    else:
		        print("request_common_neurons_command not  existed")
				
		def filesystem_test2():
		    _thread.stack_size(1024 * 6)
		    _thread.start_new_thread(filesystem_test, ())
		
		filesystem_test1()		
		# filesystem_test2() 

if I execute filesystem_test1(), it ran well, but if I execute filesystem_test2(), system resatrt by watchdog

	[0;32mI (1328) spiram: SPI SRAM memory test OK[0m
	[0;32mI (1328) heap_init: Initializing. RAM available for dynamic allocation:[0m
	[0;32mI (1328) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
	[0;32mI (1334) heap_init: At 3FFC5AC8 len 0001A538 (105 KiB): DRAM[0m
	[0;32mI (1341) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
	[0;32mI (1347) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
	[0;32mI (1354) heap_init: At 40095524 len 0000AADC (42 KiB): IRAM[0m
	[0;32mI (1360) cpu_start: Pro cpu start user code[0m
	[0;32mI (1365) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
	[0;32mI (1373) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
	[0;32mI (52) cpu_start: Starting scheduler on PRO CPU.[0m
	mp_task_heap_size 2883540
	execute default
	MicroPython v1.9.4-166-gd61d119c-dirty on 2018-06-29; ESP32 module with ESP32
	Type "help()" for more information.
	>>> filesystem_test
	ets Jun  8 2016 00:22:57
	
	rst:0x8 (TG1WDT_SYS_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0xee
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:2
	load:0x3fff0018,len:4
	load:0x3fff001c,len:4572

By the way, I have just compiled the official project with SPIRAM enabled, no interfering codes added.  
is there any ideas about this issue?
Thank you!

robert-hh, 2018-06-29:
> _thread.start_new_thread(filesystem_test, ())

What is filesystem_test() doing. Is that identical to filesystem_test1()?

FFtust, 2018-06-29:
Hi, @robert-hh ,
    I made a mistake, there is no filesystem_test(),  it should be 
    "_thread.start_new_thread(filesystem_test1, ())",  not ""_thread.start_new_thread(filesystem_test, ())

robert-hh, 2018-06-29:
The same code on Pyboard stops with the error:
MemoryError: memory allocation failed, allocating 6144 bytes
The line which flags the error is the one which starts the thread. The cause might be the same, just the error behaviour different.

dpgeorge, 2018-06-30:
Thanks for the report.  Currently, ESP32 with external SPI RAM and threading is not supported.  It may work in some cases but not all, because having the TCB and stack in external SPI RAM can often lead to contention of the external SPI bus.  In the test above, it's probably because filesystem access, which uses the external SPI flash, conflicts with external SPI RAM.

To fix it a few things need to be done: TCP *and* stack for the new thread need to be in internal RAM; the GC must be careful to trace the stack of the thread (it does do this automatically when the thread's stack lives in the GC heap).

FFtust, 2018-07-02:
I got it, and thans for your respond!