micropython/micropython #3908: Would it be possible to publish the binaries as github releases?

jhgoodwin, 2018-07-01:
Hi,

While I appreciate the desire to have people build the code from source, the effort to get a build environment up is fairly steep.

For two days now, I've been chasing down various dependencies to get a unix build for a mips processor. I'm playing with this project https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks

Is it possible to just publish a set of the binaries so people can just make use of the project?

Alternately, would it be possible to create a docker container or something which defines the proper build environment(s) so that people can just run a simple command and remove all doubts for how to configure a build environment?

Thanks,
John

dpgeorge, 2018-07-02:
> the effort to get a build environment up is fairly steep

It depends on the target arch.  For the unix port it should be as simple as installing gcc and make and running "make" in the ports/unix directory (and maybe disabling extra features in mpconfigport.mk).

> I've been chasing down various dependencies to get a unix build for a mips processor.

Installing a MIPS gcc should be enough.  What are the exact issues you have?

> Is it possible to just publish a set of the binaries so people can just make use of the project?

I'm not sure exactly what you mean by "binaries" here, but we do publish firmware builds here: https://micropython.org/download/

jhgoodwin, 2018-07-02:
So, on OSX, I tried to pull it down and build it, but then it seemed like I needed to run:

```
brew install automake
```

But it seemed I didn't have libffi, so I pulled the submodules, then went into that folder, but when trying to follow the readme, it says I need to install more stuff:
```
brew install libtool
```
So, I get the ./autogen.sh working, but running ./configure I have no idea what to supply, so I try:
```
./configure --build=x86_64 --host=mips-gnu-linux
```

So I may or may not have gotten it to build using:
```
PKG_CONFIG_PATH=../../lib/libffi/mips-gnu-linux-gnu/ make axtls
```

Then I had to copy the ffi.h and ffitarget.h from  into the unix/ports then run make agin
Then now I get ssgidefs.h missing error.

I guess I'm saying that it seems like I'm missing something since it doesn't seem like it should probably be this hard.

Regarding the binaries, I appreciate the firmware builds being available. That's great for popular platforms which needs the entire firmware. I was just suggesting that if there's a supported binary format available, it could be prebuilt and available for download. For example, I didn't see any of the firmware binaries being for any MIPS platforms (which seems to be what the ports/unix would support).

I hope I don't come across as unappreciative. Just seems like I'm missing something major on trying to benefit from the hard work that went into this project.

Thanks, John

dpgeorge, 2018-07-02:
> But it seemed I didn't have libffi, so I pulled the submodules,

You can build the unix port without needing any of the submodules or any additional libraries if you edit `ports/unix/mpconfigport.h` and set to 0 all the config variables, except for `MICROPY_USE_READLINE` which should stay as 1.

> I was just suggesting that if there's a supported binary format available, it could be prebuilt and available for download. For example, I didn't see any of the firmware binaries being for any MIPS platforms

There are currently no binary builds available for the unix port.  There was some discussion about this in #1295 but that's a low priority item.  And MIPS builds would be something extra to add to that.

You may find the "cross" branch useful because it has some additional commits (from long ago, but may still work) to build on MIPS: https://github.com/micropython/micropython/commits/cross

jhgoodwin, 2018-07-02:
Thanks for the help. Setting the configs to 0 didn't seem to help. I still ended up with this:
```
GEN build/genhdr/qstr.i.last
In file included from modffi.c:32:
In file included from ./ffi.h:67:
./ffitarget.h:45:11: fatal error: 'sgidefs.h' file not found
# include <sgidefs.h>
          ^~~~~~~~~~~
1 error generated.
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build/genhdr/qstr.i.last'
```
I may check out the other stuff later. At this point, I think I'll have to wait a bit. This effort was mainly to see if I could control these cameras easier now that I got access to them without the embedded software. My goal was just to get access to the video feed, learn to write the feed to sd card, learn to motion capture, then ship the motion capture to a second stage filter to remove annoying alerts like clouds or a fly in front of the camera. I thought maybe micropython might make some of that simpler. At this point, I'll write it in bash or C. Thanks anyway!

dpgeorge, 2018-07-02:
> I still ended up with this:

Please try doing a "make clean" because if `MICROPY_PY_FFI` is disabled then modffi.c should not even be included in the build.

jhgoodwin, 2018-07-02:
```
grep -r "\(1\)" mpconfigport.h
mpconfigport.h: * Copyright (c) 2013, 2014 Damien P. George
mpconfigport.h:    #define MICROPY_MAKE_POINTER_CALLABLE(p) ((void*)((mp_uint_t)(p) | 1))
mpconfigport.h:#define MICROPY_USE_READLINE_HISTORY (1)
mpconfigport.h:#define MICROPY_PY_UHASHLIB_SHA1    (0)
mpconfigport.h:#define MICROPY_FATFS_LFN_CODE_PAGE    (437) /* 1=SFN/ANSI 437=LFN/U.S.(OEM) */
mpconfigport.h:// Define to 1 to use undertested inefficient GC helper implementation
mpconfigport.h:#define MP_PLAT_PRINT_STRN(str, len) do { ssize_t ret = write(1, str, len); (void)ret; } while (0)
mpconfigport.h:// Bionic libc in Android 1.5 misses these 2 functions
mpconfigport.h:// of the fields [in struct dirent] not defined in POSIX.1 by testing whether
```
Then
```
make clean
make axtls
make
```
Still end up with the same error.

mitchins, 2018-07-21:
@jhgoodwin  @dpgeorge I have been thinking recently, and have required exactly this... in my case to unit-test micro python libraries on CI/CD (Python3.4 doesn't treat the async and other libraries the same so I wanted the real thing).

So I went out and made it... because looking around no-one has maintained an up to date set.

Here are the docker images:
https://hub.docker.com/r/mitchins/micropython-linux/tags/

There are stable and latest, each one will run on a daily cron:

Stable checks if the latest tag on GitHub for micro python is newer than what is on docker hub and builds if so, additionally tagging the 'v1.9.4' and 'stable'. (I have started at v1.9.3)
Latest will pull down the master repo and build it and push to latest.

Regardless of my current interest in the project, it'll build and be up to date on a nightly basis thanks to travis.

For those interested in the meat of it: It's based off stretch-slim, and really is pretty darn small anyway (once removing all tools and caches used in construction), after initially looked at Alpine, I really would rather not wrestle with it's headers and libraries.

Source is here https://github.com/mitchins/docker-micropython-linux

j16180339887, 2018-07-25:
@mitchins Make it into docker images is a great idea.

I think we can make some dockerfiles to build and release the binaries.

take [Archlinux](https://hub.docker.com/r/base/archlinux/) for example:

```dockerfile
RUN pacman -Syu git make pkgconf --noconfirm --needed
```

then [choose a gcc cross-compiler](https://www.archlinux.org/packages/?sort=&q=gcc), like [arm-none-eabi-gcc](https://www.archlinux.org/packages/community/x86_64/arm-none-eabi-gcc)

 ```dockerfile
RUN pacman -S arm-none-eabi-gcc --noconfirm --needed
```

Run makefile (I didn't test this code)

```sh
make CC=$CC CROSS_COMPILE=$ARM_PREFIX axtls
make
```

More gcc compilers can be found [here](https://aur.archlinux.org/packages/?O=0&K=gcc)

mitchins, 2018-07-25:
@j16180339887 the docker images are already created? have you tried pulling:

`docker run mitchins/micropython-linux:latest`
or 
`docker run mitchins/micropython-linux:stable`

it's based off Debian and uses standard gcc/python, give it a go and let me know how it goes

j16180339887, 2018-07-25:
Yeah, I can see what it does in the [dockerfile](https://github.com/mitchins/docker-micropython-linux/blob/latest/Dockerfile) 

Which builds the linux-x86_64 micropython interperter.

Cross-compiled [linux-arm](https://en.wikipedia.org/wiki/ARM_architecture) and [mips](https://en.wikipedia.org/wiki/MIPS_architecture) micropython on the other hand takes some efforts to do.

mitchins, 2018-07-25:
@j16180339887 the only other use case for docker would be arm64 which the raspberry pi supports. which to be honest, for arm64, creating .deb for the raspberry pi might be the best use of resource as it's likely the largest use case. Given that the raspberry pi can run gcc, we can actually compile it on the device itself so we don't need the cross compiler, however automating this would be more challenging. 
Having said that, and done a quick google... it may be possible to use emu + arm64 on travis to do the builds: https://github.com/rycus86/docker-arm64v8-alpine-qemu

The docker image above, will at least let people experiment with micro python or use it on services such as travis to run test phases.

j16180339887, 2018-07-25:
Qemu is a workaround for cross-compiling 

But it will defeat the purpose of the cross-compiler.

I could just make one single dockerfile which compiles multi-architecture micropython binaries at one time, if I can figure out :)

mitchins, 2018-07-25:
@j16180339887 love the idea, I've devoted about as much time as I have for the cause with the docker images, at least it's a start and someone can experiment or run micro python on travis docker stage... but I've love to see this progress. Let me know if you get this working, I'll be working on some micro python based libraries...

j16180339887, 2018-07-25:
@mitchins ok then, keep in touch :)

For anyone struggling with compiling micropython

actually Alpine Linux, [the most lightweight linux distro](https://hub.docker.com/r/library/alpine/tags/), has pre-built micropython packages

It can found [here](https://pkgs.alpinelinux.org/packages?name=micropython*&branch=edge)

Install it like this

```dockerfile
apk add micropython micropython-cross
```

j16180339887, 2018-07-31:
@mitchins I have successfully cross compiled micropython using aarch64-gcc (arm64)

I use [archlinux image](https://hub.docker.com/r/base/archlinux/) (actually I don't really use archlinux at all), then

```sh
pacman -S git make gcc pkgconf autoconf automake --noconfirm
pacman -S aarch64-linux-gnu-gcc --noconfirm

git clone --depth 1 https://github.com/libffi/libffi $HOME/libffi
git clone --depth 1 https://github.com/micropython/micropython $HOME/micropython

# mpy-cross doesn't need to use cross compiler
cd $HOME/micropython/mpy-cross
make clean
make

export ARM_PREFIX="aarch64-linux-gnu"
export LIB_CACHE=$HOME/$ARM_PREFIX
export CROSS_COMPILE="$ARM_PREFIX-"
export CROSS_COMPILER_PREFIX="$ARM_PREFIX-"
export CFLAGS_EXTRA=""
export LDFLAGS_EXTRA="-static -L$LIB_CACHE/lib64 -L$LIB_CACHE/lib"
cd $HOME/libffi
make clean
./autogen.sh
./configure --prefix=$LIB_CACHE --host=$ARM_PREFIX
make
make install

cd $HOME/micropython/ports/unix
make clean
make V=1 CROSS_COMPILE="$CROSS_COMPILE" axtls
make V=1 CROSS_COMPILE="$CROSS_COMPILE"
./micropython # Runs on arm64 Linux only
```

### TODO
* Linux
  * arm32
  * arm64 (done!)
  * x86
  * x64
* Android
  * arm32
  * arm64
  * x86
  * x64
* Make it into one Dockerfile

mitchins, 2018-07-31:
@j16180339887 that's awesome. What do you think about creating .deb's for use with raspbian? This is probably the most popular arm64 system that people will be using.

j16180339887, 2018-07-31:
I don't know how to package it into a .deb file

but it is static compiled binary so just

```
sudo cp micropython /usr/bin/
sudo chmod +x /usr/bin/micropython
```

Or just

```sh
sudo install micropython /usr/bin/micropython
```

then it should work, you don't need to install libffi or other dependencies.

j16180339887, 2018-08-06:
[micropython-aarch64-linux-gnu.zip](https://github.com/micropython/micropython/files/2261060/micropython-aarch64-linux-gnu.zip)
[micropython-arm-linux-gnueabi.zip](https://github.com/micropython/micropython/files/2261061/micropython-arm-linux-gnueabi.zip)
[micropython-mips-linux-gnu.zip](https://github.com/micropython/micropython/files/2261062/micropython-mips-linux-gnu.zip)
[micropython-x86_64-w64-mingw32.zip](https://github.com/micropython/micropython/files/2261069/micropython-x86_64-w64-mingw32.zip)

Ok, I found that debian got a lot of more cross-compiler packages

```sh
gcc-aarch64-linux-gnu
gcc-alpha-linux-gnu
gcc-arm-linux-gnueabi
gcc-arm-linux-gnueabihf
gcc-hppa-linux-gnu
gcc-hppa64-linux-gnu
gcc-i686-linux-gnu
gcc-m68k-linux-gnu
gcc-mips-linux-gnu
gcc-mips64-linux-gnuabi64
gcc-mips64el-linux-gnuabi64
gcc-mipsel-linux-gnu
gcc-powerpc-linux-gnu
gcc-powerpc-linux-gnuspe
gcc-powerpc64-linux-gnu
gcc-powerpc64le-linux-gnu
gcc-riscv64-linux-gnu
gcc-s390x-linux-gnu
gcc-sh4-linux-gnu
gcc-sparc64-linux-gnu
gcc-x86-64-linux-gnux32
gcc-mingw-w64-i686
gcc-mingw-w64-x86-64
```

Take `gcc-aarch64-linux-gnu` `gcc-arm-linux-gnueabi` `gcc-mips-linux-gnu` `gcc-mingw-w64-x86-64` for example
```sh
apt install -y git make gcc pkgconf autoconf automake libtool-bin texinfo python

git clone --depth 1 https://github.com/jeremyhu/libffi $HOME/libffi
git clone --depth 1 https://github.com/micropython/micropython $HOME/micropython

# mpy-cross doesn't need to use cross compiler
cd $HOME/micropython/mpy-cross
make clean
make

################# arm64 Linux begin #####################
apt install -y gcc-aarch64-linux-gnu

export CC_PREFIX="aarch64-linux-gnu"
export LIB_CACHE=$HOME/$CC_PREFIX
export CROSS_COMPILE="$CC_PREFIX-"
export CROSS_COMPILER_PREFIX="$CC_PREFIX-"
export CFLAGS_EXTRA="-I$LIB_CACHE/include"
export LDFLAGS_EXTRA="-static -lffi -L$LIB_CACHE/lib64 -L$LIB_CACHE/lib"
cd $HOME/libffi
make clean
./autogen.sh
./configure --prefix=$LIB_CACHE --host=$CC_PREFIX
make
make install

cd $HOME/micropython/ports/unix
make clean
make V=1 CROSS_COMPILE="$CROSS_COMPILE" axtls
make V=1 CROSS_COMPILE="$CROSS_COMPILE" LDFLAGS_EXTRA="$LDFLAGS_EXTRA"
cp micropython $HOME/$CC_PREFIX

################# arm64 Linux end   #####################
################# arm32 Linux begin #####################
apt install -y gcc-arm-linux-gnueabi

export CC_PREFIX="arm-linux-gnueabi"
export LIB_CACHE=$HOME/$CC_PREFIX
export CROSS_COMPILE="$CC_PREFIX-"
export CROSS_COMPILER_PREFIX="$CC_PREFIX-"
export CFLAGS_EXTRA="-I$LIB_CACHE/include"
export LDFLAGS_EXTRA="-static -lffi -L$LIB_CACHE/lib64 -L$LIB_CACHE/lib"
cd $HOME/libffi
make clean
./autogen.sh
./configure --prefix=$LIB_CACHE --host=$CC_PREFIX
make
make install

cd $HOME/micropython/ports/unix
make clean
make V=1 CROSS_COMPILE="$CROSS_COMPILE" axtls
make V=1 CROSS_COMPILE="$CROSS_COMPILE" LDFLAGS_EXTRA="$LDFLAGS_EXTRA"
cp micropython $HOME/$CC_PREFIX

################# arm32 Linux end   #####################
################# mips Linux begin #####################
apt install -y gcc-mips-linux-gnu

export CC_PREFIX="mips-linux-gnu"
export LIB_CACHE=$HOME/$CC_PREFIX
export CROSS_COMPILE="$CC_PREFIX-"
export CROSS_COMPILER_PREFIX="$CC_PREFIX-"
export CFLAGS_EXTRA="-I$LIB_CACHE/include"
export LDFLAGS_EXTRA="-static -lffi -L$LIB_CACHE/lib64 -L$LIB_CACHE/lib"
cd $HOME/libffi
make clean
./autogen.sh
./configure --prefix=$LIB_CACHE --host=$CC_PREFIX
make
make install

cd $HOME/micropython/ports/unix
make clean
make V=1 CROSS_COMPILE="$CROSS_COMPILE" axtls
make V=1 CROSS_COMPILE="$CROSS_COMPILE" LDFLAGS_EXTRA="$LDFLAGS_EXTRA"
cp micropython $HOME/$CC_PREFIX

################# mips Linux end   #####################
################# win64 begin #####################
apt install -y gcc-mingw-w64-x86-64
#cp /usr/arm-linux-gnueabi/include/malloc.h /usr/arm-linux-gnueabi/include/alloca.h
export CC_PREFIX="x86_64-w64-mingw32"
export LIB_CACHE=$HOME/$CC_PREFIX
export CROSS_COMPILE="$CC_PREFIX-"
export CROSS_COMPILER_PREFIX="$CC_PREFIX-"
export CFLAGS_EXTRA="-I$LIB_CACHE/include"
export LDFLAGS_EXTRA="-static -lffi -L$LIB_CACHE/lib64 -L$LIB_CACHE/lib"
cd $HOME/libffi
make clean
./autogen.sh
./configure --prefix=$LIB_CACHE --host=$CC_PREFIX
make
make install

cd $HOME/micropython/ports/windows
make clean
make V=1 CROSS_COMPILE="$CROSS_COMPILE" axtls
make V=1 CROSS_COMPILE="$CROSS_COMPILE" LDFLAGS_EXTRA="$LDFLAGS_EXTRA"
cp micropython.exe $HOME/$CC_PREFIX
################# win64 end #####################
```

However, I got some errors when compile micropython statically using Android NDK.

@dpgeorge @jhgoodwin @mitchins What do you think about [github binary release](https://help.github.com/articles/distributing-large-binaries/) using dockerfiles.

mitchins, 2018-08-06:
@j16180339887 thanks for this, interesting.
I think that the Dockerfile solves a particular issue of either people wanting to get their feet where without much knowledge or wanting to run anywhere through docker. 
Additionally, it allows travis or other CI to use it as mentioned above.

.deb packages could be distributed through bintray for raspberry pi or desktop users, and I think would be the long term goal... but we cannot solve every linux/*bsd user and so Docker offers a great catch all as well.

j16180339887, 2018-08-06:
@mitchins @jhgoodwin @dpgeorge 

I will send a PR contains a dockerfile as soon as possible,

even though I can not successfully cross-compile Android version.

I will target these platform for the first release:
```sh
arm32-linux
arm64-linux
mips-linux
mips64-linux
x86-linux
x64-linux
win32 (no libffi)
win64 (no libffi)
```

`.deb` file I will try to figure it out after the first release, maybe with your help, @mitchins :)

jonnor, 2023-07-30:
Proposing to rename ths issue to something like "Publish binaries for Unix/Windows ports".

mattytrentini, 2023-07-31:
I think we can safely close this issue; a lot of time has passed and building has changed somewhat.

closed 2023-07-31 by mattytrentini