micropython/micropython #3913: ESP32: isconnected() and status() methods do not work

dreasGh, 2018-07-02:
Hi,

i have pull the repository a couple of days ago and now i'm on the commit d8dc918d (py/compile: Handle return/break/continue correctly in async with.) of 23/06/2018.
Now when the ESP32 disconnect for some reasons in STA_IF mode, isconnected() method always returns True and the ESP32 log shows as follows:

`I (5476584) wifi: STA_DISCONNECTED, reason:201`

If i call the status() method he always returns None so i'm not able to determinate the status of the network.STA_IF.

Thanks in advance for the help.

dpgeorge, 2018-07-03:
Thanks for the report.  Indeed, in some cases `isconnected()` will return `True` if the connection to the AP is lost.  The logic is that it will remain in the "connected" state if it is trying to reconnect to the AP.

But probably it would make sense for `isconnected()` to return `True` only if it is fully connected, with an IP address, and ready to send/receive IP frames.

mitchins, 2018-07-09:
This makes sense, that it should only be True if status == 'STAT_GOT_IP'.

As a side note, I also noted the issue that status is None sometimes, is this expected or something that also needs looking into? 
(This may have even been occurring when I was fully connected with IP address, which may complicate the above issue)

dpgeorge, 2018-07-09:
> As a side note, I also noted the issue that status is None sometimes, is this expected or something that also needs looking into?

Link status is not yet implemented: `wlan.status()` should return an integer describing the current status of the link/connection.  For esp8266 it returns things like idle, connecting, no-ap-found, etc (these are constants in the `network` module).

mitchins, 2018-07-10:
@dpgeorge can you please take a quick look at commit 70d197c in my branch. I need to do some extended testing however it builds ok. My biggest concern is the confusion I encountered around the statics table. I used the ESP8266 for inspiration but had to insert the enum that would otherwise have been inside user_interface.h.

dpgeorge, 2018-07-11:
I left comments on your commit on your branch.

commit 385fa518 (2018-07-14): esp32: Implement WLAN.status() return codes.

Resolves #3913: missing esp32 status() implementation.
closed 2018-07-14 by dpgeorge