micropython/micropython #3917: [NUCLEO_F091RC] ADC reading of internal sensors

rolandvs, 2018-07-02:
While trying the pyb.ADCAll() function I came across a possible bug:
Given:
```
import pyb
adc=pyb.ADCAll(12, 0x70000)
print(adc.read_core_temp())
```
Will print the temperature of the chip around 36.35 degrees while writing this. Whenever another core function is read, reading the temperature again will show false readings like 73.358 degrees.
Using a soft reset (ctrl-D) will not help to get the first situation back, a hard reset is necessary.

dpgeorge, 2018-07-03:
Yes I did notice the temp reading was not working in some cases on this board.  I don't know the exact reason, it needs some investigation.

rolandvs, 2018-07-05:
Have tested It again and it seems that the first time it is working, however if you have used `read_core_vref()` subsequent readings will give a false value only to cleared when a Power On Reset is given (ctrl-D will not work).

dpgeorge, 2018-08-04:
This should be fixed by 7be5bb367212b6949e74f73d90af01f8b68f1352

closed 2018-08-04 by dpgeorge

rolandvs, 2018-08-04:
Verified and working!