micropython/micropython #3918: `ustruct.unpack_from` `offset` arg can't be a keyword

dhalbert, 2018-07-02:
The documentation for `ustruct.unpack_from(fmt, data, offset=0)` shows the `offset` argument as being optional, with a default value: http://docs.micropython.org/en/latest/pyboard/library/ustruct.html?highlight=unpack_from#ustruct.unpack_from. This is the same as in CPython. But you can't name the arg:
```
MicroPython v1.9.4 on 2018-05-11; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import ustruct
>>> ustruct.unpack_from('b', b'abc', offset=2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
```
My impression is that, like other builtins that might take keyword args, you don't want to bother checking, for space reasons. So this should perhaps just be documented if the implementation of `unpack_from` is not going to change.

dpgeorge, 2018-07-03:
> My impression is that, like other builtins that might take keyword args, you don't want to bother checking, for space reasons.

Yes that's correct.

> So this should perhaps just be documented if the implementation of unpack_from is not going to change.

Sounds good.  Because none of the functions in `ustruct` take keyword arguments I suggest to add a small note about this at the start of the page for this module, to cover all functions.

robhamerling, 2019-07-10:
Why does the Micropython doc of the ustruct library mentions:
       ustruct.unpack_from(fmt, data, offset=0)
while Micropython (1.11-142) refuses 'offset' as keyword argument?

jason-neal, 2020-01-04:
> Why does the Micropython doc of the ustruct library mentions:
ustruct.unpack_from(fmt, data, offset=0)
while Micropython (1.11-142) refuses 'offset' as keyword argument?

To preserve code space in the implementation it is not a keyword argument but a positional argument (with a default value if left out). 
In #5196 it was proposed to start using the positional-only syntax from [PEP 570](https://www.python.org/dev/peps/pep-0570/) in the MicroPython docs to clarify this.

dpgeorge, 2020-01-06:
Would be good to use `/` notation here.

dpgeorge, 2020-01-12:
The '/' notation was added to the docs in 7ef2f65114f092be6303c145a2560fdf522dcde0

closed 2020-01-12 by dpgeorge