micropython/micropython #3919: [NUCLEO_F091RC] pyb.ADCAll(12,mask) some mask values freeze system

rolandvs, 2018-07-03:
Maybe related to #3917, also when using `pyb.ADCAll(12,MASK)` some masks will freeze micropython (at least REPL does not respond anymore and reset is the only thing to get out of it).

Instances that will freeze the system: `pyb.ADCAll(12)` and a mask value between `0x7FFF4` and `0x7FFFF`.

dpgeorge, 2018-07-03:
> also when using pyb.ADCAll(12,MASK) some masks will freeze micropython

I think that's just because it disables the UART on PA2/PA3.

rolandvs, 2018-07-03:
Aha that seems logical. Should there be a check for this?

dpgeorge, 2018-07-04:
> Should there be a check for this?

It's not easy to do it in a smart way... and maybe the user does actually want to turn all pins to ADC.  Probably it's best to address this in the docs for ADCAll.

rolandvs, 2018-07-05:
Documenting is one thing, but would it be wise to set the default mask not to (-1) but to the 0x700000 value for the internal sensors. In most other cases one would assign the analog channels per channel and not all of them at once.

dpgeorge, 2018-07-06:
> but would it be wise to set the default mask not to (-1) but to the 0x700000 value for the internal sensors

That would be wise, but here we are trying to maintain existing behaviour, which for `ADCAll` was to put *all* pins in to ADC mode.

I'd like to revamp the `ADC` class in general, make it more consistent across ports, and also introduce something like `ADCBlock` which would allow to control an ADC peripheral block in a more user-friendly way, rather than the ad-hoc methods in `pyb.ADCAll`.

rolandvs, 2018-07-06:
Ok, I cannot image why you would enable all adc channels at once as its use is in most cases very specific. Also earlier documentation imho showed that the example that fiddles directly with the registers was favoured as it did not have the side effects of ADCAll :-). I will use the mask and will PR the doc...

rolandvs, 2018-08-04:
With the exception of disabling the UART when all GPIO is set to analog input it is well documented.

closed 2018-08-04 by rolandvs