micropython/micropython #3924: Generators have wrong type compared to CPython

ryannathans, 2018-07-04:
CPython

```
>>> def this_is_a_generator():
...   yield
...   return
...
>>> def this_is_a_function():
...   return
...
>>> type(this_is_a_generator)
<class 'function'>
>>> type(this_is_a_function)
<class 'function'>
>>> a = this_is_a_generator()
>>> type(a)
<class 'generator'>
```

Micropython

```
>>> def this_is_a_generator():
...     yield
...     return
...
>>> def this_is_a_function():
...     return
...
...
>>> type(this_is_a_generator)
<class 'generator'>
>>> type(this_is_a_function)
<class 'function'>
>>> a = this_is_a_generator()
>>> type(a)
<class 'generator'>
```

type(this_is_a_generator) returns <class 'generator'> in micropython where this should be <class 'function'>

dpgeorge, 2018-07-04:
Thanks for the report.  Indeed, a generating function that is not yet started does have a different type to a normal function.  I'm not sure what the Python "specs" say about this, whether it's required to have the same type.  Why exactly does it matter what the type is?

ryannathans, 2018-07-04:
@dpgeorge I was writing cross-platform code, specifically and simply, a function that does something with an argument depending on if its type is a function or not.

On micropython the code fails when a generator is passed in because the type of the argument passed in is not a function, but is instead a generator type (which should be what the function returns, not what the function is).

dpgeorge, 2018-07-04:
> a function that does something with an argument depending on if its type is a function or not

Ok.  But note that built-in functions also have a different type to user-defined (Python) functions, for both CPython and uPy.  You might want to use `callable(f)` to test all these cases, including non-started generators.

dpgeorge, 2018-07-04:
> it just takes not to rely on CPython's idiosyncrasies, where it happens that 2 different objects have the same type, while other implementations have different types.

And in this vein note that uPy has many different types of built-in functions, the type of functions with 1 arg is different from one with 2 args, eg: `type(struct.calcsize) is type(struct.pack)` is False.

It is possible to "fix" the issue here and make non-started generators have the same type as a function: it would be done by inlining the `objgenerator.c:gen_wrap_call()` code into `objfun.c:fun_bc_call()` and detecting in `fun_bc_call()` if a function is a generator or not before executing it.  That may actually decrease code size but would make all bytecode function calls a little less efficient due to the check for a generator (which is currently done at compile time).

peterhinch, 2018-07-04:
You can use `isinstance` portably to distinguish functions, generators and generator functions.
```python
type_genf = type((lambda: (yield)))  # Type of a generator function
type_gen = type((lambda: (yield))())  # Type of a generator
```

jonnor, 2024-09-29:
There seems to be a portable solution for this problem, in the comment above. Can we close this issue?

closed 2025-02-04 by projectgus