micropython/micropython #3927: change `type name[]` to `type * name` inside structs

adritium, 2018-07-05:
In C, it's [not possible to completely create and initialize a struct-type containing flexible array member](https://stackoverflow.com/questions/8687671/how-to-initialize-a-structure-with-flexible-array-member)

```C
typedef struct _mp_obj_tuple_t {
    mp_obj_base_t base;
    size_t len;
    mp_obj_t items[];
} mp_obj_tuple_t;
```

To see it in action, [run this online code snippet](https://onlinegdb.com/Hy1D5nifX)

```C
#include <stdio.h>

typedef struct _mp_obj_tuple_t {
    int base;
    int len;
    int *items; // change this to int items[] and you'll get a compile error
} mp_obj_tuple_t;

int main()
{
    int * p = ((int []){1, 2, 3});
    
    void * parent = &((mp_obj_tuple_t){
        1,
        2,
        (int[]){99, 88, 77, 66}
    });
    
    mp_obj_tuple_t * obj =  (mp_obj_tuple_t *) parent;
    
    int i;
    for(i = 0; i < 3; i++)
    {
        printf("%d, %d\n", p[i], obj->items[i]);
    }

    return 0;
}
```

The benefit to using `mp_obj_t items[];` is that it's contiguous in memory with rest of struct members.
Is that fact used anywhere?

The cost is not being able to perform initialization of this type _unless you run additional code_ to initialize that member.

dhylands, 2018-07-05:
If you really want to initialize a static object with a flexible array, you can do somethning like this:
```
typedef struct _mp_obj_tuple_t {
    int base;
    int len;
    int items[];
} mp_obj_tuple_t;

typedef struct {
    mp_obj_tuple_t tuple_base;
    int tuple_items[3];
} specific_t;

int main()
{
    int * p = ((int []){1, 2, 3});

    void * parent = &((specific_t){
        {1, 2},
        {99, 88, 77}
    });
    
    mp_obj_tuple_t * obj =  (mp_obj_tuple_t *) parent;
    
    int i;
    for(i = 0; i < 3; i++)
    {
        printf("%d, %d\n", p[i], obj->items[i]);
    }

    return 0;
}
```
(available here: https://onlinegdb.com/B19R1Asz7)

adritium, 2018-07-05:
Yes that's true but I want the uPy code base to be changed.

adritium, 2018-07-05:
Ok I got it: if we change `type items[]` to `type * items` you won't be able to allocate it using `m_new_obj_var`

closed 2018-07-05 by adritium