micropython/micropython #3930: RFC: Standardised "wait for interrupt" API

nickovs, 2018-07-07:
Right now the stm32 and teensy ports have a board-specific call to drop the processor to a low-power mode and wait until some sort of interrupt happens in the background. This is very useful for a whole host of functions, notably when you have some  interrupt code that needs to return with low-latency but you also want that code to kick into life some foreground processing. In the absence of full-fledged semaphores the foreground code can loop waiting for any interrupt to happen and then check for a flag set by the interrupt hander.

I would like to propose that this project adds `wfi()` to the `machine` module and standardise it's API. Right now the pyboard version takes no parameters; I think it might be helpful to add an optional timeout argument. This is not _strictly_ necessary since this can always be faked by setting a hardware timer and the timer interrupt will cause `wfi()` to return, but that eats up a precious hardware timer so it would be nice to avoid it. That said, many CPUs that target embedded systems have a low-level CPU instruction to achieve WFI functionality deep in the hardware, and these instructions don't take a timeout, so I would be OK with not having a timeout if others think it would be tricky on some ports. If we do have a timeout then it should probably be in milliseconds since that seems to be the standard for these things and maps to a fairly useful range.

If it makes this proposal any more appealing I'm willing to write an ESP32 implementation.

Comments and suggestions welcomed!

dpgeorge, 2018-07-08:
> This is very useful for a whole host of functions

Yes I agree.

> In the absence of full-fledged semaphores the foreground code can loop waiting for any interrupt to happen and then check for a flag set by the interrupt hander.

You do need to be careful though because it may be that the IRQ arrives between checking the flag and doing the "wfi", which may leave you waiting for an IRQ that already fired.  For example see this commit: c0496fd44de562cd11e66acd9e42f796c3dcb5fb and also similar fed1b4fb56421e8f8a1ee54b2c16296204176bb3

nickovs, 2018-07-08:
@dpgeorge You make a great case for implementing full-fledge semaphores! 

I would happily replace this suggestion with a suggestion for binary semaphores. If we are going to do that we might as well implement counting semaphores too (since the code is mostly the same).

dpgeorge, 2018-07-09:
> I would happily replace this suggestion with a suggestion for binary semaphores.

Let's think if there are existing things that can already do this:
- Polling objects with the select module may work because this allows to "sleep" until something becomes readable/writable.  User file objects (derived from `io.IOBase`) could act as the "semaphore" object.  The main thread could wait on a user file object, and the IRQ could "signal" the file object by setting a flag in it.  It's then up to the port to make polling work efficiently.
- Using threads: `_thread.allocate_lock()` makes a binary semaphore which the main thread can "acquire" and then the IRQ can "release".  (The CPython `threading` module has a full `Semaphore` object which is built from the primitive `_thread.allocate_lock`.)

Both of these options may already work on certain ports.  You don't need to use multiple threads to use the lock object in the `_thread` module.  In fact, even if a port doesn't have multi-threading capabilities, we could still give it a `_thread` module with a simple lock primitive so it can handle signalling from IRQs.

dhylands, 2018-07-09:
> You do need to be careful though because it may be that the IRQ arrives between checking the flag and doing the "wfi", which may leave you waiting for an IRQ that already fired.

The correct way to handle this is to mask interrupts, do your check and then do the WFI call (with interrupts masked). The WFI call returns whether interrupts are masked or not. Typically you unmask interrupts right after the WFI call, and the interrupt handler will run at that time.

Reference: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0360e/CHDEEIJB.html in particular WFI will suspend until "an interrupt masked by PRIMASK becomes pending".

This is exactly how the linux kernel does it as well (on the Cortex-A series of processors).

dpgeorge, 2018-07-10:
> The correct way to handle this is to mask interrupts ...

Yeah, that's how it's now being done in the SPI and SD card driver (see commits linked above).

nickovs, 2018-07-10:
> The correct way to handle this is to mask interrupts, do your check and then do the WFI call

Which takes us back to my original proposal, which is to standardise an API to expose a WFI instruction! We could always put some rubric and example code in the documentation for the call to make sure that people understand the need to do their checks in the correct sequence.

dpgeorge, 2018-07-11:
> Which takes us back to my original proposal, which is to standardise an API to expose a WFI instruction

Actually, I just remembered that `machine.idle()` is pretty much exactly this: http://docs.micropython.org/en/latest/pyboard/library/machine.html?highlight=machine%20idle#machine.idle .  You can see that esp8266 implements it, and esp32 also does but uses FreeRTOS yield.

dpgeorge, 2018-12-04:
`machine.idle()` seems fit for this purpose, so closing.

closed 2018-12-04 by dpgeorge