micropython/micropython #3931: unix port of uos.ilistdir reports the incorrect file type

spon-ww, 2018-07-09:
The documenttation for the uos.ilistdir function (https://docs.micropython.org/en/latest/pyboard/library/uos.html#filesystem-access) says:

uos.ilistdir([dir])
This function returns an iterator which then yields tuples corresponding to the entries in the directory that it is listing. With no argument it lists the current directory, otherwise it lists the directory given by dir.

The tuples have the form (name, type, inode[, size]):

name is a string (or bytes if dir is a bytes object) and is the name of the entry;
type is an integer that specifies the type of the entry, with 0x4000 for directories and 0x8000 for regular files;
inode is an integer corresponding to the inode of the file, and may be 0 for filesystems that don’t have such a notion.
Some platforms may return a 4-tuple that includes the entry’s size. For file entries, size is an integer representing the size of the file or -1 if unknown. Its meaning is currently undefined for directory entries.

However, at least on the unix port, the type field of the tuple is 4 for directories and 8 for regular files:

spon@ajatar:unix$ ./micropython 
MicroPython v1.9.4-189-g34344a4 on 2018-06-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import uos
>>> for x in uos.ilistdir():
...     print(x)
...     
...     
... 
('unix_mphal.c', 8, 10627372)
('mpconfigport.mk', 8, 10627362)
('modusocket.c', 8, 10627360)
('moduos_vfs.c', 8, 10627358)
('gccollect.c', 8, 10627345)
('coverage-frzstr', 4, 10627335)

I'm not sure if this is an issue in uos, the unix port of uos, or the documentation.

pfalcon, 2018-07-09:
It's a bug in unix port. The idea is to have compatibility with stat.S_IFREG, stat.S_IFDIR flags. So, it should be:

~~~
t->items[1] = MP_OBJ_NEW_SMALL_INT(dirent->d_type << 12);
~~~

(Likewise, in vfs_posix.c instead of adhoc if-else ladder.)

pfalcon, 2018-07-09:
> cannot always

There's a standard API and standard value used in MicroPython, for any port. If a characteristic if at all available (anything can be not available anywhere), it uses the standard value.

dpgeorge, 2018-07-10:
Thanks for the report.  Indeed it looks like an oversight in unix's uos module, that it returns the `d_type` value instead of the value that would correspond to calling `stat` on the entry.

For portability, the conversion from d-type to stat type should use `DTTOIF()` if it's available, otherwise explicitly check and covert each known value.

dpgeorge, 2018-07-11:
Should be fixed by 3ab2f3fb2b807a49cf4835cfff7b5e5139a1da76

closed 2018-07-11 by dpgeorge

spon-ww, 2018-07-11:
I confirm that as of3ab2f3f, the type value returned by ilistdir now matches the docs on the unix port.

Fantastic work - thanks!