micropython/micropython #3932: ESP32: Build issue

PikWay, 2018-07-09:
Hello,

I've came back after couple months of not building my own firmware and having problem.
- Newest microptyhon repo
- Matching esp-idf 
- Built mpy-cross 

But firmware build ends with: 
LINK build/bootloader.elf
Create build/bootloader.bin
esptool.py v2.1
Create build/partitions.bin
make: *** No rule to make target `lib/berkeley-db-1.xx/btree/bt_close.c', needed by `build/genhdr/qstr.i.last'.  Stop.

I'm searching for some docs describing how to prepare build env, but no luck.

dhylands, 2018-07-09:
Did you do `git submodule update --init` and anytime I see qstr errors I normally do a clean and then a rebuild.

dpgeorge, 2018-07-10:
You probably also need to run `git submodule update --init` in the ESP IDF repository after checking out the relevant git hash.

PikWay, 2018-07-10:
Thanks! It works,

closed 2018-07-10 by PikWay