micropython/micropython #3934: RFC: having a standard way to specify UART buffer lengths

dpgeorge, 2018-07-10:
Configuring a UART read/write buffer length is important, to trade between memory usage and ability to buffer a lot of data.  In some cases you really want a large buffer, and in other cases a small one will be fine.  So it makes sense to be able to dynamically configure it.

The stm32 port allows changing the read buffer length via the constructor like `UART(1, 9600, read_buf_len=200)`  Other ports don't currently support this and it would be good to standardise this capability (see eg esp32 request: #3765).

Since stm32 already hase `read_buf_len` it might be good to use this for the other ports.  Then the writing buffer would be configured using `write_buf_len`.

But, it may also be a good chance to pick better (maybe?) names.  For example shorter names like `rx_buf_len`,`tx_buf_len`.  Or even just `rxbuf` and `txbuf`.  The latter names are a bit more general and could possibly support passing a preallocated buffer (eg a bytearray) as well as an integer.  They could also find (re)use on other peripherals that may need to configure tx/rx buffers.

So the question is: should we stick with `read_buf_len` and introduce `write_buf_len`, or move to something like `rxbuf` and `txbuf`?  (In the latter case, stm32 would retain `read_buf_len` for compatibility, at least for now.)

Of course, this assumes that such a feature (configuring buffer length) is useful and should be introduced as a general feature of the UART class, which I think it should.

peterhinch, 2018-07-10:
> Of course, this assumes that such a feature (configuring buffer length) is useful and should be introduced as a general feature of the UART class, which I think it should.

Most definitely.

dpgeorge, 2018-12-05:
Implementations of `rxbuf` (and `txbuf` where appropriate) when done for stm32, esp32 and esp8266 in 8007d0bd16e1007453c7c801345458db5d663e21, 9ddc182ec7195722de426d5461ed1d70d9aedf7f and 52bec93755e70dc2f5bea00377190b2278954c78 respectively.

closed 2018-12-05 by dpgeorge