micropython/micropython #3936: ESP8266-only crash on 1.9.4

garthk, 2018-07-11:
I have some code that doesn't crash on OSX, but does on ESP8266 (D1 Mini).

`repro.py`:
```py
#!/usr/bin/env micropython

SAMPLES = [32767, 32687, 32448, 32050, 31497, 30790, 29934, 28931, 27787, 26509, 25100, 23570]
SAMPLES = SAMPLES[:1] # increase if you don't crash; it's fiddly

@micropython.native
def DAC4():
    err = 0
    output = []
    for samp in SAMPLES:
        samp = samp + 32768 - err
        if samp > 65535:
            samp = 65535
        elif samp < 0:
            samp = 0
        output.append(samp >> 11)
        err = samp & 0x7ff
    return output

print(DAC4())
```

MicroPython v1.9.4 on 2018-05-11; darwin version:
```
$ micropython repro.py 
[31]
```

Pasted into raw mode on the ESP8266:
```
MicroPython v1.9.4-8-ga9a3caad0 on 2018-05-11; ESP module with ESP8266
Type "help()" for more information.
>>> 
raw REPL; CTRL-B to exit
�OKERROR: xtensa bcc out of range
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 31108, room 16 
tail 4
chksum 0x28
load 0x3ffe8000, len 1100, room 4 
tail 8
chksum 0x4e
load 0x3ffe8450, len 3268, room 0 
tail 4
chksum 0x09
csum 0x09
```

I get similar crashes with `ampy run`, only wrapped up in an `ampy.pyboard.PyboardError`.

What extra steps can I take to help isolate the problem?

dpgeorge, 2018-07-11:
Thanks for the report.  There's enough information here to know the cause: the built-in xtensa assembler is not able to handle conditional jumps which jump more than 8 (signed) bits away, see https://github.com/micropython/micropython/blob/master/py/asmxtensa.c#L134-L136

dpgeorge, 2018-07-12:
Should be fixed by 8c9c167dc6986a44acf9f255ff96e6822165e3e0 (actually that patch is more of a workaround than a true fix because it removes the need for long conditional branches, instead of adding support for long conditional branches on xtensa).

closed 2018-07-12 by dpgeorge