micropython/micropython #3939: Create function to initialize mp_obj_module_t

adritium, 2018-07-11:
`mp_obj_t mp_obj_new_module(qstr module_name)` creates _and_ adds a module to `mp_loaded_modules_dict`.

https://github.com/micropython/micropython/blob/d974ee1c2fc3256ab367eeeb020921f8a5b0dd61/py/objmodule.c#L101-L116

There is no function similar to `mp_obj_new_list` that simply creates `mp_obj_module_t` object and initializes it.

stinos, 2018-07-12:
Arguably the issue should be the other way around: mp_obj_new_module should just create the module and after that mp_module_register should be used explicitly if desired (like when importing): it feels wrong that creating a new module object without importing it exposes it in sys.modules automatically. Also imo returning an existing object in a function called new_module feels a bit awkward, and the functionality of getting an existing module (possible to add members to it) is readily available via mp_module_get anyway.

adritium, 2018-07-12:
That would break existing code that assumes `mp_obj_new_module` works the way it currently does.

stinos, 2018-07-12:
Off course, the question is if anybody out there relies on that. Not sure if there's a compelling reason to have a module which isn't even loaded in sys.modules? I looked around and actually this is a reported problem, see https://github.com/micropython/micropython/issues/967

adritium, 2018-07-13:
> the question is if anybody out there relies on that

That's impossible to know so not a valid basis to introduce a breaking change (in general).

> it feels wrong that creating a new module object without importing it exposes it in sys.modules automatically

The unofficial convention of every other **mp_obj_new_XYZ()** is _"create a new mp_obj_XYZ_t"_ but what `mp_obj_new_module` actually means is _"create a new top-level Python module."_ 

So now I do think a breaking change should be introduced because `mp_obj_new_module` breaks the unofficial convention mentioned above.

Following renames should occur:

1. `mp_obj_new_module` -> `mp_obj_new_py_module`
2. `mp_module_get` -> `mp_py_module_get`
3. `mp_module_register` -> `mp_py_module_register`

All these functions access `MP_STATE_VM(mp_loaded_modules_dict)` which means they reference top-level Python modules.

New functions to create:

1. `mp_obj_new_module` - creates and initializes `mp_obj_module_t`

stinos, 2018-07-13:
> That's impossible to know so not a valid basis to introduce a breaking change (in general).

Yeah but I didn't mean that should be taken as the reason to break or not; the reason in my opinion is that mp_obj_new_module does more than just creating the module and makes the task of mp_py_module_register less obvious

> "create a new top-level Python module."

mp_obj_new_module is used for all modules, including those which aren't top-level?

> Following renames should occur:

```mp_py_``` isn't used anywhere so introducing it just for 3 functions which already have a clear name derives from conventions and seems unnecessary.

> mp_module_del_entry - deletes entry

What do you mean with 'entry'?

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge