micropython/micropython #3941: Having issues building the firmware for STM32F4

HaidStefan, 2018-07-12:
Hello,

I have tried to build the firmware on the current master branch for the STM32F4 discovery board. My goal is to extend the firmware with some custom c-libraries, but at first i wanted to try to just build and flash the unchanged firmware. 

I cloned the git repo and used the `git submodule update --init` command as described in some tutorials. 

First i tried to build it on Windows 10 with CMD using GNU Tools Arm Embedded Toolchain. 
To build the Firmware i used the command: `make BOARD=STM32F4DISC`
but i receive the following error: 

```
LINK mpy-cross
strip: 'mpy-cross': No such file
make[1]: *** [../py/mkrules.mk:134: mpy-cross] Error 1
make[1]: Leaving directory 'C:/workdir/github_python/micropython/mpy-cross'
make: *** [../../py/mkrules.mk:108: ../../mpy-cross/mpy-cross] Error 2
```

I searched a lot but sadly could not find a working solution. 

The second try was using MINGW64. I have installed the latest MSYS2and updated everything using
 `pacman -Syu` .

After i wanted to install the mingw-w64-arm-none-eabi-gcc package but with `pacman -S mingw-w64-arm-none-eabi-gcc` it didn't work. 
 
Does anyone know a possible solution for my issues? My next try would be to set up a vm with linux...
I am thankful for every help because I really want to try to extend the firmware... 

many thanks
cheers

Stefan

chrismas9, 2018-07-12:
For Windows 10 WSL bash shell works well. With the latest WIN10 update you can choose from several distributions. I use Ubuntu and follow the README instructions for building under Linux

mikewadsten, 2018-07-12:
I think you generally need to compile `mpy-cross` first. Try this:

```
# from the micropython/ root directory
$ make -C mpy-cross
$ make -C ports/stm32 BOARD=STM32F4DISC
```

HaidStefan, 2018-07-13:
When executing the command `make -C mpy-cross` i get the following error:
```
c:\workdir\github_python\test\micropython (master -> origin)
λ make -C mpy-cross
make: Entering directory 'c:/workdir/github_python/test/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
...
...
...
CC gccollect.c
CC ../ports/windows/fmode.c
LINK mpy-cross
strip: 'mpy-cross': No such file
make: *** [../py/mkrules.mk:134: mpy-cross] Error 1
make: Leaving directory 'c:/workdir/github_python/test/micropython/mpy-cross'
```

I just cloned the git repo and didnt change any files.

robert-hh, 2018-07-13:
strip: 'mpy-cross': No such file

That's a typical error on windows. The script wants to strip the final binary mpy-cross instead of mpy-cross.exe. As long as the exe file still exists, you can ignore the error.

HaidStefan, 2018-07-13:
@robert-hh but when i build it for the STM32board i get a second error (see my first post).

robert-hh, 2018-07-13:
The second error looks like following the first one.
You could go into the mpy-cross/Makefile, line 17, and define PROG = mpy-cross.exe

HaidStefan, 2018-07-13:
With this fix, mpy-cross builds successfully now, thanks!

Unfortunately i get another error now when executing `make -C ports/stm32 BOARD=STM32F4DISC`

Error: 
```
make[1]: Leaving directory 'C:/workdir/github_python/test/micropython/mpy-cross'
MPY modules/dht.py
Traceback (most recent call last):
  File "modules/dht.py", line 1
SyntaxError: invalid syntax
make: *** [../../py/mkrules.mk:118: build-STM32F4DISC/frozen_mpy/dht.mpy] Error 1
make: Leaving directory 'C:/workdir/github_python/test/micropython/ports/stm32'
```

dpgeorge, 2018-07-13:
> Unfortunately i get another error now when executing...

This is because Windows doesn't support symbolic links.

The easiest way around these issues is to run: `make -C ports/stm32 BOARD=STM32F4DISC FROZEN_MPY_DIR=` (be sure to do a `make -C ports/stm32 BOARD=STM32F4DISC clean` first though).

HaidStefan, 2018-07-13:
I managed to build it successful now. 
It is also important that the path to the arm-toolchain has no spaces in it. (See #3900)

Thanks a lot for everyone helping!

closed 2018-07-13 by HaidStefan