micropython/micropython #3943: RFC: refine and standardise machine.ADC class

dpgeorge, 2018-07-13:
Reading ADC values is pretty common and it would be great if all ports had the same behaviour in this respect.  The current situation is:

- For stm32 and esp32:
```python
adc = machine.ADC(pin) # pin is any pin object
adc.read() # returns value between 0 and 4095
```

- For esp8266:
```python
adc = machine.ADC(id) # id=0 or 1
adc.read() # returns value between 0 and 1024
```

- For cc3200:
```python
periph = machine.ADC()
channel = periph.channel(pin=id) # id is a pin id, a string
channel.value() # returns value between 0 and 4095
```

To at least make reading values standard across ports, the return values should be in the same range across ports.  I also think it's a good idea to make explicit the units/range in the name of the read method, because then it's completely unambiguous as to what it returns.

My proposal would be to specify the behaviour of `machine.ADC` to at least support the following usage:
```python
adc = machine.ADC(id) # id can be an integer, pin, or any other identifier
adc.read_u16() # returns a raw integer reading between 0 and 65535 inclusive
adc.read_mv() # returns a calibrated voltage reading in millivolts (int, could be negative)
adc.read_v() # returns a calibrated voltage reading (int, or float if port supports floats)
```

The `read_u16()` method is there to get full access to the raw underlying value.  `read_mv()` is more than just convenience: on systems that are able to calibrate the value it is very useful to be able to read a calibrated value (functionality not directly available with `read_u16()`), and it's an integer to avoid the need for floats.  `read_v()` is convenience and would be `read_mv()/1000`.

The advantage of using a suffix for the units is that the existing `read()` method on the various ports can stay for backwards compatibility (at least for now).

Then there are a few items that would be useful to configure, but I don't have a concrete proposal on how to configure these just yet:
- setting bit-width precision of the ADC peripheral
- setting sample time for the reading

rolandvs, 2018-07-13:
IMHO - Isn't this a big limitation introduced here. The ADC systems of the three architectures are completely different. Rather than making it standard across ports, would we not benefit if the functionality of those systems have being freed from the dudgeons of register settings and be made available at the surface?

Having calibrated values that can be read back would help, however using different adc_vref, scaling factor, gain, etc. would be parameters you would like use/add to make this worthwhile. Example: if I measure a voltage or current or whatever, having mV instead of ticks, would still require to add code to account for a voltage divider and/or shunt that is used to connect to the ADC.

peterhinch, 2018-07-13:
I think calculations around external components are best done in the application. Shunt or voltage dividers might be electronically selectable. A programmable bias voltage might be applied via a DAC. In trying to generalise, where do you stop?

Compare with a physical instrument which doesn't usually try to second-guess your measurement setup.

chrismas9, 2018-07-13:
I agree real world conversions are too varied to standardise, but left justified 16 bit, mV and V would be good. There are more and more micros offering 14bit and 16bit ADC's (sometimes by oversampling) and this proposal provides consistent methods for all precisions up to 16bit.
> read_v() is convenience and would be read_mv()/1000

This will loose precision. 1mV in 3.3V is less than 12 bit. Any 14 or 16bit ADC would be reduced to less than 12bit resolution. It would be better to derive `read_v()` directly from` read_u16()` using floating point scale factor.

> adc.read_mv() # returns a calibrated voltage reading in millivolts (int, could be negative)

`could be negative` It would be useful to be able to specify an offset in the constructor, eg 0V or midpoint. For AC coupled audio you would specify midpoint or 1.65V to get signed numbers. Otherwise I don't see how you could get signed numbers except on the few micros with differential analog inputs.

To support 24bit delta sigma converters in the future it may be worth reserving ` read_u24()` and ` read_uV()` as integer methods.

Quite a few micros have more than one ADC. There are several cases where it would be useful to specify which ADC to use on a pin. Example1: @peterhinch is trying to do simultaneous sampling to measure phase between two inputs. This would be more accurate if each signal was measured simultaneously on a different ADC. Example 2: ADC1 is being used to stream audio or dat logging. Periodically you want to measure battery voltage with ADC2 so as not to disrupt the stream from ADC1.

This could be done in the pin id (eg X1.ADC2) or with a kwarg (ADC_ch=2). There may be a better way but it would be good to ensure that the new API can support these sorts of extensions without breaking backward compatibility.

> setting bit-width precision of the ADC peripheral

Some micros achieve higher precision by oversampling and decimation (ATSAMD51 can stretch its 12bit ADC to 16bit). Should the raw precision and oversampling be specified separately (ADC_prec=12, ADC_ovrsmpl=256) or combined into one value (ADC_prec=16)? Oversampling relies on there being guassian noise in the signal. It could be misleading to just specify 16bit precision when 4bits come from oversampling because with a quiet signal you might still only get 12bit precision.

There is discussion elsewhere (eg #3854) about interrupts. The ATSAMD51 has a Window Monitor that can generate interrupts if the ADC value crosses an upper or lower limit. Very useful for fast fault detection (eg motor overcurrent). With 256x oversampling conversions are slow. End of conversion interrupt would be useful. Maybe somthing worth allowing for in the future.

chrismas9, 2018-07-14:
I'm not suggesting implementing advanced features now. But by discussing them now I hope we can make the API open to future enhancements without breaking backwards compatibility. Elsewhere there is discussion on enhanced precision for the timer API and to some extent this is being influenced by the existing API. The STM32H7 has 16bit ADCs with sub millivolt precision so don't define 'read_V ()' with 1mV precision, etc.

I felt that during the early development of the 'machine' API that it should only support the lowest feature set available on all micros. Hence no support for hardware tristate, reading pin values of outputs, etc. I am hoping that future port specific enhancements can be added in a backwards compatible way so users can exploit the capabilities of some parts.

dpgeorge, 2018-07-15:
> read_v() is convenience and would be read_mv()/1000
> 
> This will loose precision.... It would be better to derive read_v() directly fromread_u16() using floating point scale factor.

Yes, very good point!  Thanks for that.

> could be negative: It would be useful to be able to specify an offset in the constructor, eg 0V or midpoint.... Otherwise I don't see how you could get signed numbers except on the few micros with differential analog inputs.

My aim with the wording here was just to leave it open to the possibility of negative values, instead of restricting it.  So we could perhaps introduce offsets in the future, or the user could wrap an ADC instance in a custom object that shifted the value, and still provide a valid `read_mv()` method.

> To support 24bit delta sigma converters in the future it may be worth reserving read_u24() and read_uV() as integer methods.

Yes, exactly.  The idea with having the units in the suffix of the name allows to extend precision/range further if/when needed, following a consistent pattern.  This also follows how `time.ticks_ms()`, `time.ticks_us()` work.

The alternative is something like what is being discussed with the Timer period, to specify the units with an argument.  Eg `adc.read(mult=1000)` would return millivolts, and `adc.read(mult=1000000)` would return microvolts.  Although for efficiency it probably wouldn't support keyword args so it would look like `adc.read(1000)` and `adc.read(1000000)`, which isn't very readable.  Also, reading values is very common (unlike setting a Timer period) so that's why I think it makes sense to provide short and fast methods like `read_v()` and `read_mv()`.

dpgeorge, 2018-07-15:
> Quite a few micros have more than one ADC
> ...
> [Selecting the ADC peripheral] could be done in the pin id (eg X1.ADC2) or with a kwarg (ADC_ch=2). There may be a better way but it would be good to ensure that the new API can support these sorts of extensions without breaking backward compatibility.

I agree that there should be scope to support multiple ADC peripherals.  But doing an ADC read on a pin is the simple and common thing to do, so I think should be simple in the way you write the code to do it.  More complex operations would be more complex to write.

The way the cc3200 does it (see at the top) is that you create an ADC peripheral object, then create a channel associated with a pin, then read that channel.  I prefer how it's done by the other ports: just create an ADC "channel" directly, which can be read straightaway.

If you need finer control over the ADC peripheral itself (rather than just the channel/pin) I think that should be done via another class like `ADCBlock`.  So the `ADC` class would represent the concept of an endpoint/channel connected to some analog signal, which can be sampled.  And the `ADCBlock` class would represent the internal ADC peripheral block which controls the sampling.  This way you don't need to worry about the ADC block if you don't want to, and ports don't need to expose this functionality if they don't have it.  `ADCBlock` could have a factory function (eg `channel()`) which produces `ADC` objects that are explicitly associated with that ADC block.

dpgeorge, 2018-07-15:
> Rather than making it standard across ports, would we not benefit if the functionality of those systems have being freed from the dudgeons of register settings and be made available at the surface?

The idea is to (at least attempt to!) do both: have a simple but usable base that is the same across all ports, that can be relied upon, and then extend that base with more advanced features using additional methods/classes (eg `ADCBlock`, as discussed above).  The basic functionality of reading an ADC value is still going to be needed even if you also want to do other, more advanced things.

To start with, the aim here is at least to make the ADC reading methods have the same return values.  Otherwise it's confusing going from port to port.

> Example: if I measure a voltage or current or whatever, having mV instead of ticks, would still require to add code to account for a voltage divider and/or shunt that is used to connect to the ADC

It's easy to add such functionality by wrapping an ADC object in a custom object that does the needed transformation.  But transforming a raw 12-bit value to Volts is not as easy because it requires calibration, and that is provided by some ADC's internally.

jonathanhogg, 2021-09-08:
Hey @dpgeorge!

Is this idea still active / on the roadmap? I have need of calibrated voltage readings for my current project so I've hacked up a simple addition to the ESP32 port to add `read_v()` and `read_mv()` methods that use the IDF ADC calibration API (pulls coefficients from the eFuse of the device).

The code is pretty grim at the moment, but would look better refactored to match your `ADCBlock()`. So if it's something that's still on the cards, I'm happy to rework my code and just implement the #4213 API in full.

dpgeorge, 2021-09-13:
@jonathanhogg yes this idea is still on the roadmap.  Calibrated ADC voltage readings are very useful, and non-trivial to do the calibration for some parts (eg STM32, ESP32), so having it all done by a single method would be very welcome.

I think #4213 is pretty close but could do with some tweaks, eg specify just `read_uv()` (to return an integer being microvolts), and not `read_v()` nor `read_mv()`.

jonathanhogg, 2021-09-13:
OK, I'm going to do some more work on the ESP32 code and submit a PR. I've already decided that I can't live with the hack I've written and as I need calibrated readings – like the counter stuff – I'm committed to supporting it regardless.

@dpgeorge: if you've got any further thoughts on what the right API should be then lemme have 'em.

dpgeorge, 2022-01-21:
The improved ADC API including specification of ADCBlock was done in 4d2f487ee1ac93098b90c3108b071f6e855c9e7f.  So I'll close this issue, the original reason for opening it is now addressed.

closed 2022-01-21 by dpgeorge