micropython/micropython #3946: post flash AP_IF behavior not consistent with documentation on ESP32

ezeeetm, 2018-07-14:
From the docs:

> After a fresh install and boot the device configures itself as a WiFi access point (AP) that you can connect to. The ESSID is of the form MicroPython-xxxxxx where the x’s are replaced with part of the MAC address of your device (so will be the same everytime, and most likely different for all ESP8266 chips). The password for the WiFi is micropythoN (note the upper-case N). Its IP address will be 192.168.4.1 once you connect to its network. 

However, a fresh install on an ESP32 does not enable the AP _at all_.

Manually enabling the AP works with the following:
```
>>> ap_if = network.WLAN(network.AP_IF)
>>> ap_if.active(True)
```

This does bring up the AP to which you can connect, but:
-the ESSID is not MicroPython-xxxxxx , it is ESP_085559 (anybody know where 085559 comes from?  ITs not a MAC address or part of a MAC address)
- authentication is open (no password required)

peterhinch, 2018-07-14:
Unfortunately there isn't (yet) a section in the docs for ESP32. I think you've been reading the ESP8266 docs: the ESP8266 hardware stores connection details whereas that on the ESP32 does not.

jonnor, 2024-09-29:
ESP8266 is the special one here. All other ports one has to do this manually. The docs for ESP32 are at https://docs.micropython.org/en/latest/esp32/quickref.html#networking

closed 2024-09-29 by jonnor