micropython/micropython #3951: ports/windows/msvc/genhdr.targets: enable generic compile command

adritium, 2018-07-17:
Some companies build **using** Visual Studio build system **with** GNU toolchain.

Yes, I know.

I propose modifying **ports/windows/msvc/genhdr.targets** so that instead of calling `cl` when the compiler is needed,

https://github.com/micropython/micropython/blob/46091b8a9572c67b817aeea1f9c4c39dc41e6aac/ports/windows/msvc/genhdr.targets#L88

it instead calls `<CompilerExe>` which would be defined as 
`<CompilerExe>$(GCCPath)\bin\arm-none-eabi-gcc.exe</CompilerExe>` 
somewhere else.

stinos, 2018-07-18:
Interesting; does MicroPython build with this change?

Note the canonical way of doing this is using the ```ClToolExe``` property since that's also used by msbuild itself.

adritium, 2018-07-18:
> Interesting; does MicroPython build with this change?

Why wouldn't micropython build with `<CompilerExe>$(GCCPath)\bin\arm-none-eabi-gcc.exe</CompilerExe>`?

> Note the canonical way of doing this is using the `ClToolExe` property since that's also used by msbuild itself.

That term doesn't appear as a properly documented term on MSDN: the search `ClToolExe site:https://msdn.microsoft.com/` yields only blog and forum posts.
Therefore, I don't consider it to be supported even if it works.

stinos, 2018-07-18:
> Why wouldn't micropython build with <CompilerExe>$(GCCPath)\bin\arm-none-eabi-gcc.exe</CompilerExe>?

I didn't really mean the whole of MicroPython but the msvc port. Or is that not what you're trying to do here? It selects files which shouldn't be used when using gcc but only when using the msvc toolset (might still work though, don't know, that's why I'm asking), hence the seperate msvc directory. 

> Therefore, I don't consider it to be supported even if it works.

Faire enough but then I wonder how you're going to instruct the rest of the build to not use cl.exe but something else?

adritium, 2018-07-18:
I am using ONE compiler for the ENTIRE build.
The micropython code is in a separate `.vcxproj` but all `.vcxproj` have the line `<CompilerExe>$(GCCPath)\bin\arm-none-eabi-gcc.exe</CompilerExe>`

adritium, 2018-07-18:
> A standard vcxproj file doesn't know that property so won't use it

I see your point and the issue.

We have custom `.Targets` file, which is a `MSBuild` XML file, and this contains the recipe which uses `<CompilerExe>`

So I see what I want can't be generalized without using some built-in **compiler-switcher** command.