micropython/micropython #3961: ESP8266 crashes on wake from deepsleep

peterhinch, 2018-07-19:
This was reported [on the forum](https://forum.micropython.org/viewtopic.php?f=16&t=5000).

If the following code is pasted at the REPL the board wakes after 10s but crashes: there is no response to the REPL or to pings, and the red LED on the reference board is lit.
```python
import machine
rtc = machine.RTC()
rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)
rtc.alarm(machine.RTC.ALARM0, 10000)
machine.deepsleep()
machine.reset()  # Crashes with or without this
```
A second issue is that the `handler` kwarg to rtc.irq is rejected: 
```python
rtc.irq(trigger=rtc.ALARM0, handler=lambda *_ : None, wake=machine.DEEPSLEEP)
```
produces `TypeError: extra keyword arguments given`.

Hardware: Adafruit Feather Huzzah, build:
```
MicroPython v1.9.4-227-gab02abe-dirty on 2018-07-19; ESP module with ESP8266
```

dpgeorge, 2018-07-19:
Thanks for reporting.  I'm pretty sure you need to connect pin 16 to the reset pin (RST) to be able to wake from deep sleep.  On the ESP8266, pin 16 is connected to the RTC domain and must be externally wired to RST to fully reset the rest of the system (other MCUs would do this internally).  I tried this and it works with the above code.

`handler` can't really be supported for a wake from deep sleep because all RAM is reset, there's no way to "remember" any callbacks to execute.

peterhinch, 2018-07-20:
Thanks for that - I've quoted this as a response to the forum query.

However the argument handling of `rtc.irq` seems wrong and fails even with the default `wake=machine.IDLE`:
```python
import machine
rtc = machine.RTC()
rtc.irq(trigger=rtc.ALARM0, handler=lambda *_ : None)
```
results in:
```python
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
TypeError: extra keyword arguments given
>>> 
```

dpgeorge, 2018-12-04:
> However the argument handling of `rtc.irq` seems wrong

At the moment the esp8266 doesn't support the `hardler` argument to `rtc.irq()`, so that's the remaining issue here.

projectgus, 2024-08-28:
Closing as the remaining part is tracked in #2701

closed 2024-08-28 by projectgus