micropython/micropython #3963: esp8266 crashes using ussl with uselect.poll

neurino, 2018-07-19:
Hello,

please try this code:

```
import socket
import select
import ssl

proto, dummy, host, path = 'https://httpbin.org/post'.split('/', 3)
port = 443 if proto is 'https:' else 80        
ai = socket.getaddrinfo(host, port, 0, socket.SOCK_STREAM)[0]
method = 'POST'

print(proto, host, path, port, method)

s = socket.socket(ai[0], ai[1], ai[2])
s.connect(ai[-1])

if proto == "https:":
    s = ssl.wrap_socket(s, server_hostname=host)

p = select.poll()
p.register(s, select.POLLOUT)

go = True
while go:
    for o, e in p.ipoll(250):
        print(o, e)
        if e & select.POLLOUT:
            print('writing')
            s.write(b'{0} /{1} HTTP/1.0\r\n'.format(method, path))
            s.write(b'Host: {0}\r\n'.format(host))
            s.write(b'Accept: application/json\r\n')
            s.write(b'\r\n')
            s.write(b'\r\n')
            p.modify(s, select.POLLIN)
        elif e & select.POLLIN:
            print('reading')
            s.read()
            s.close()
            p.unregister(s)
            go = False

```

which on first run results in

```
https: httpbin.org post 443 POST
Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
OSError: [Errno 22] EINVAL
```

while running again ends in fatal exception:

```
https: httpbin.org post 443 POST
Fatal exception 29(StoreProhibitedCause):
epc1=0x4026c55c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00001440, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30944, room 16 
tail 0
chksum 0x92
load 0x3ffe8000, len 1084, room 8 
tail 4
chksum 0x66
load 0x3ffe8440, len 3252, room 4 
tail 0
chksum 0xc6
csum 0xc6
[gibberish...]
```

In http it runs fine — change first line of code above to

`https://httpbin.org/post'.split('/', 3)`

and you get the expected result:

```
http: httpbin.org post 80 POST
<socket state=2 timeout=-1 incoming=0 off=0> 4
writing
21
19
26
2
2
<socket state=2 timeout=-1 incoming=3fff9c60 off=0> 1
reading
b'HTTP/1.1 200 OK\r\nConnection: close\r\nServer: gunicorn/19.8.1\r\nDate: Thu, 19 Jul 2018 22:27:01 GMT\r\nContent-Type: application/json\r\nContent-Length: 212\r\nAccess-Control-Allow-Origin: *\r\nAccess-Control-Allow-Credentials: true\r\nVia: 1.1 vegur\r\n\r\n{"args":{},"data":"","files":{},"form":{},"headers":{"Accept":"application/json","Connection":"close","Content-Length":"0","Host":"httpbin.org"},"json":null,"origin":"93.66.65.6","url":"http://httpbin.org/post"}\n'

```

I'm running this build `esp8266-20180707-v1.9.4-244-g106e5945.bin` because the stable `esp8266-20180511-v1.9.4.bin` does not allow me to detach repl from UART with `uos.dupterm(None, 1)`

My target is to send data to a REST API while polling at the same time UART for RFID reader data. [My device is this](https://it.aliexpress.com/item/NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-flash-USB-serial-CP2102/32779738528.html?spm=a2g0y.10010108.1000016.1.6d1468a3dKYjq5&isOrigTitle=true)

[image]

dpgeorge, 2018-07-20:
Thanks for the report.  I could reproduce the problems.

For the crashing when run a second time: this was because lwIP was calling a TCP callback on a socket that did no longer exist (it was executing it after a soft reset, on a socket that was made before soft reset).  This was fixed in 4a2051eec7f60c4a05822da540f763c5bc1775b1

For the case of polling ussl objects: this is covered by the existing issue #3394.  I've now implemented that in 7a67f057d74e9a16e7ae7bc562e592335145eaee, and so you should be able to poll ussl objects (on both esp8266 and esp32).

closed 2018-07-20 by dpgeorge

neurino, 2018-07-24:
Thank you, I confirm it now runs in https too