micropython/micropython #3972: Problem compiling micropython for TTGO (ESP32 + Lora)

fandres, 2018-07-22:
Hi I'm trying to compile Micropython for development board OGTT
I have the following error:
/home/vagrant/esp-idf//components/bootloader_support/include/esp_flash_partitions.h:25:36: error: 'CONFIG_PARTITION_TABLE_OFFSET' undeclared (first use in this function)
[**Gist**](https://gist.github.com/fandres/95d0ce03e5c2c0abd017a94dec5e8ba8)

makefile:
ESPIDF = $(HOME)/esp-idf/
PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

include Makefile

xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/vagrant/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu
...
Thread model: single
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a)

Thank you, have a nice day

dpgeorge, 2018-07-23:
Please make sure you have checked out the correct version of the ESP IDF (and do git submodule update).  When you run "make" in the ports/esp32/ directory it will warn if the version is not correct.

fandres, 2018-07-26:
It is correct, the version is not the expected one. How can I know which version is appropriate?
[Error](https://gist.github.com/fandres/94878fc34c714be6522f66424d159c37)

Tnks

fandres, 2018-07-26:
The version gives you the error. I already compile it
Thank you so much @dpgeorge

closed 2018-07-27 by dpgeorge