micropython/micropython #3975: micropython doesn't handle directories

medecau, 2018-07-23:
By default python handles directories as input by looking for ```__main__.py``` but micropython fails to look it up.

```
$ micropython ./
OSError: [Errno 21] EISDIR
$ python ./
Hello world!
$ ls
__main__.py
```

I guess the interesting stuff would then happen when zipfile module becomes a real thing. So one could start looking into using PEX with micropython. 😉

dpgeorge, 2021-05-25:
It's possible to use `-m` to get the same behaviour:
```
$ micropython -m ./
Hello world!
$ ls
__main__.py
```
So I'll close this as "won't implement, has an alternative".

closed 2021-05-25 by dpgeorge

medecau, 2021-07-21:
The behavior for importing a module is not the same as "executing" a dir.
```
$ ls && cat __*
__init__.py __main__.py
print(__file__)
print(__file__)
$
$ micropython -m ./
.//__init__.py
.//__main__.py
```
I can work with this though.
Thank you @dpgeorge .