micropython/micropython #3976: Compilation randomly fails when compiling fast (race condition)

ryannathans, 2018-07-24:
Frequently when compiling (stm32 port, but also sometimes mpy-cross), I experience failures such as 

```
from ./mpconfigport.h:37,
                 from ../../py/mpconfig.h:45,
                 from ../../py/obj.h:29,
                 from gccollect.c:30:
../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_pcd.h:123:34: fatal error: stm32f4xx_hal_pcd_ex.h: No such file or directory
 #include "stm32f4xx_hal_pcd_ex.h"
```
I am compiling in ubuntu via windows subsystem for linux. This started to occur after updating to windows 10 1803, which sped up compilation from 2.5 minutes down to 43 seconds. CPU utilisation went from ~40% avg to 100% after updating. I am building with '-j12' when calling make.

It seems that sometimes compilation races ahead instead of blocking when prerequisites haven't finished compiling?

The workaround is to clean and retry 1-5 times until compilation is successful.

Currently running 
```
OS: Ubuntu 16.04 xenial
Kernel: x86_64 Linux 4.4.0-17134-Microsoft
Shell: bash 4.3.48
CPU: Intel Core i7-6800K CPU @ 3.401GHz
RAM: 42923MiB / 65434MiB

GCC: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Make: GNU Make 4.1
```

dhalbert, 2018-07-24:
I think it would be good to file an issue in https://github.com/Microsoft/WSL/issues. This sounds like it might be a filesystem race condition or something like that.

peterhinch, 2018-07-24:
I have never seen this running Linux natively and building with `-j 8`. CPU usage goes to 100% on all cores. So I'd concur in suspecting WSL.

ryannathans, 2018-07-24:
found it: https://github.com/Microsoft/WSL/issues/3281

adritium, 2018-07-25:
@dhalbert @peterhinch I disagree. You are assuming all dependencies in the makefiles are fully specified.

There likely are under-specified dependencies that only come to light when exceeding a certain number of `make` jobs. 

Here's a simplification of the issue with two `make` targets.

```make
# Creates blah.txt
t1:
    create_file.bat > blah.txt

# Uses blah.txt
t2:
    do_some_thing.bat blah.txt
```

With `make -j1` there wouldn't be a problem because the targets would be executed in the order they appear. However, with more than one `make` job, you'll get race conditions because sometimes `t1` would get executed first and sometimes `t2` and if `t2` is first, you'd get `blah.txt does not exist`

`make` does not INFER dependencies. The correction is to set `t1` as a dependency of `t2`:

```make
# Creates blah.txt
t1:
    create_file.bat > blah.txt

# Uses blah.txt
t2: t1
    do_some_thing.bat blah.txt
```

But things are usually more complicated than this example. It's definitely possible there are file dependencies between different makefiles, not just within one makefile.

And how do even detect these file dependencies that exist? Generally speaking, for each `make` target you'd have to detect all filesystem access and then if/when another target attempts to read/write some other file created/deleted in another target, you'd have to mark that as a dependency between the two targets (if it's not already explicitly marked).

Or just demo version of [Electric Cloud](http://electric-cloud.com/get-started/) - IT'S NOT ACTUALLY A CLOUD PRODUCT - that will do all this for you AND spit it out in a nice `.xml` for you to post-process later.

dhalbert, 2018-07-25:
@adritium See https://github.com/Microsoft/WSL/issues/3281, which describes a similar issue in WSL and also references other similar WSL issues.

What you say about `make` is true but I don't think that's the problem in this case, given the circumstantial evidence of similar issues in WSL and the current structure of the MicroPython makefiles.

dhylands, 2018-07-25:
Also, the header file in question is a static file (i.e. not generated), so there are only a few reasons why it wouldn't be found:
1 - Incorrect options were passed to gcc (i.e. it wasn't looking in the right location)
2 - The header file is missing
3 - Some type of filesystem bug, make bug, gcc bug

Since compiling again makes things work, it tends to rule out 1 & 2 (given that its a static header file).

ryannathans, 2018-07-25:
@dhylands I think this makes the most sense - I was thinking it wasn't finding the compiled output but it's clearly not finding the source file

dpgeorge, 2018-08-13:
Closing because this definitely looks like a problem with the underlying OS.

closed 2018-08-13 by dpgeorge