micropython/micropython #3978: make deploy to Pyboard is broken

peterhinch, 2018-07-25:
Using firmware built from today's (25th July) code:
```shell
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/ports/stm32
$ make BOARD=PYBV11 deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Writing build-PYBV11/firmware.dfu to the board
Traceback (most recent call last):
  File "../../tools/pydfu.py", line 65, in <module>
    if 'length' in inspect.getfullargspec(usb.util.get_string).args:
AttributeError: 'module' object has no attribute 'getfullargspec'
make: *** [Makefile:465: deploy] Error 1
```
This appears to be related to a recent change to `pydfu.py`. I managed to flash the firmware with dfu-util.

I don't know if this is relevant but the script I use to get the latest source, compile mpy-cross and build the Unix build is:
```shell
cd $MPDIR
echo Working...
git checkout master
git pull origin master
git submodule update --init
cd mpy-cross
make clean
make -j 8
cd ../ports/stm32
make BOARD=PYBV11 clean
make BOARD=PYBV10 clean
make BOARD=PYBLITEV10 clean
cd ../esp8266
make clean
cd ../unix
make clean
make -j 8 axtls
# If you're going to enable deplibs: see micropython/README
#make deplibs
make -j 8
```

rolandvs, 2018-07-25:
There was an update to support the latest or current version of `pyusb`. The
```
if 'length' in inspect.getargspec(usb.util.get_string).args: 
```

replaced by 
```
if 'length' in inspect.getfullargspec(usb.util.get_string).args: 
    # PyUSB 1.0.2 depreciates getargspec() 
    # PyUSB 1.0.0.b1 has the length argument 
```

peterhinch, 2018-07-25:
Sorry if I'm being dim here but I've now done
```shell
sudo pip install --upgrade pyusb
```
with apparent success but I'm still getting the error.

robert-hh, 2018-07-25:
Hi Peter, it may be unrelated, but for me deploy only works, if I use it like:

make PYTHON=python3 BOARD=PYBV11 deploy

rolandvs, 2018-07-25:
Dear Peter,

I just tried compiling with the latest master and flashed the pboard v10 without a problem... using OSX, python3(.7.0) and pyusb 1.0.2 (using the cli command as @robert-hh just answered with)

peterhinch, 2018-07-25:
@robert-hh @rolandvs Thanks, that works.
@dpgeorge This strikes me as a significant breakage, in terms of the need to upgrade `pyusb` and to change the invocation line. I fear I won't be the only one to fail to spot it...

rolandvs, 2018-07-25:
@peterhinch I'm the source for your trouble :-) See #3955.

robert-hh, 2018-07-25:
Confirmed: that was the point when I had to change the command for deploy. You definitely owe us a drink.

peterhinch, 2018-07-25:
It should be documented, if only in a forum post.

rolandvs, 2018-07-25:
Given the geographical distances I would be happy to use paypal and send you all @dgeorge, @peterhinch and @robert-hh, a beer... For starters 🍺🍺🍺

dpgeorge, 2018-07-26:
Thanks @peterhinch for the report.  It looks like problem is that Python 2's inspect module does not have the `inspect.getfullargspec` function, but it has `getargspec`.  But then in Python 3 it's the other way around: `getargspec` is deprecated in preference to `getfullargspec`.

So `pydfu.py` needs to detect this difference and do the right thing.  Eg use `hasattr(inspect, 'getfullargspec')` to detect which version it runs under.

dpgeorge, 2018-07-27:
Should be fixed by 11a38d5dc5365c9fa1d3edc05f948e21293ce127

closed 2018-07-27 by dpgeorge