micropython/micropython #3979: 'string' in mp_type_module - make it iterable

adritium, 2018-07-25:
I don't know if this will break anything else but would it make sense to enable user to type `'string' in module` where `module` is `mp_type_module`?

I know I can do `'string' in dir(module)`.

Seems like a straightforward use-case: I want to see if this string is in this module.

dpgeorge, 2018-07-26:
> 'string' in module

This is not supported by CPython so isn't part of the language, so I don't see any reason for uPy to support it.  Surely there are other ways to do what you want?

adritium, 2018-07-26:
I wanna see if a string is part of a module. 

I can do it using `’string’ in dir(my_module)`. Is that the most Pythonic way?

dhylands, 2018-07-26:
Wouldn't `hasattr(my_module, 'string')` do what you want?

adritium, 2018-07-27:
I guess.

closed 2018-07-27 by adritium