micropython/micropython #3980: esp

FFtust, 2018-07-26:


dpgeorge, 2018-07-27:
> mp_thread_deinit() will never return

Did you come across a case of this happening when running code?  If so could you share that code so the error can be reproduced?

FFtust, 2018-07-27:
HI @dpgeorge ,  
  I modify the mp_task like this,

    void mp_task(void *pvParameter) {
	    volatile uint32_t sp = (uint32_t)get_sp();
	    #if MICROPY_PY_THREAD
	    mp_thread_init(&mp_task_stack[0], MP_TASK_STACK_LEN);
	    #endif
	    uart_init();
	
	    // Allocate the uPy heap using malloc and get the largest available region
	    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
	    void *mp_task_heap = malloc(mp_task_heap_size);
	
	    soft_reset:
	    // initialise the stack pointer for the main thread
	    mp_stack_set_top((void *)sp);
	    mp_stack_set_limit(MP_TASK_STACK_SIZE - 1024);
	    gc_init(mp_task_heap, mp_task_heap + mp_task_heap_size);
	    mp_init();
	    mp_obj_list_init(mp_sys_path, 0);
	    mp_obj_list_append(mp_sys_path, MP_OBJ_NEW_QSTR(MP_QSTR_));
	    mp_obj_list_append(mp_sys_path, MP_OBJ_NEW_QSTR(MP_QSTR__slash_lib));
	    mp_obj_list_init(mp_sys_argv, 0);
	    readline_init0();
	
	    // initialise peripherals
	    machine_pins_init();
	
	    // run boot-up scripts
	    pyexec_frozen_module("_boot.py");
	    pyexec_file("boot.py");
	    if (pyexec_mode_kind == PYEXEC_MODE_FRIENDLY_REPL) {
	        pyexec_file("main.py");
	    }
	    
	    vTaskDelay(200);
	    // for (;;) {
	    //     if (pyexec_mode_kind == PYEXEC_MODE_RAW_REPL) {
	    //         vprintf_like_t vprintf_log = esp_log_set_vprintf(vprintf_null);
	    //         if (pyexec_raw_repl() != 0) {
	    //             break;
	    //         }
	    //         esp_log_set_vprintf(vprintf_log);
	    //     } else {
	    //         if (pyexec_friendly_repl() != 0) {
	    //             break;
	    //         }
	    //     }
	    // }
	    printf("***** mp_thread_deinit start****\n");
	    #if MICROPY_PY_THREAD
	    mp_thread_deinit();
	    #endif
	    printf("***** mp_thread_deinit end ****\n");
	    gc_sweep_all();
	
	    mp_hal_stdout_tx_str("PYB: soft reboot\r\n");
	
	    // deinitialise peripherals
	    machine_pins_deinit();
	    usocket_events_deinit();
	
	    mp_deinit();
	    fflush(stdout);
	    goto soft_reset;
    }

and create a thread in "_boot.py",  

	import gc
	import uos
	from flashbdev import bdev
	
	try:
	    if bdev:
	        uos.mount(bdev, '/')
	except OSError:
	    import inisetup
	    vfs = inisetup.setup()
	
	gc.collect()
	
	import _thread
	import time
	def test_func():
	    while True:
	        print("_thread test")
	        time.sleep(0.5)
	
	print("start test thread")
	_thread.start_new_thread(test_func, ())

here is the log,

	[0;32mI (381) cpu_start: Single core mode[0m
	[0;32mI (381) heap_init: Initializing. RAM available for dynamic allocation:[0m
	[0;32mI (385) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
	[0;32mI (391) heap_init: At 3FFC4E90 len 0001B170 (108 KiB): DRAM[0m
	[0;32mI (397) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
	[0;32mI (403) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
	[0;32mI (410) heap_init: At 40091B5C len 0000E4A4 (57 KiB): IRAM[0m
	[0;32mI (416) cpu_start: Pro cpu start user code[0m
	[0;32mI (210) cpu_start: Starting scheduler on PRO CPU.[0m
	start test thread
	OSError: [Errno 2] ENOENT
	***** mp_thread_deinit start****

it never run to "printf("***** mp_thread_deinit end ****\n");"  

I have traced the execution information by printing logs, and found "vTaskDelete(th->id)" was the reason, last comment has described the details.