micropython/micropython #3981: cannot create an array of fixed size

adritium, 2018-07-26:
I need to create an array that can never be allowed to increase or decrease in size.

Without creating a new type, this cannot be done.

dmazzella, 2018-07-26:
you can use a deque:

```
from ucollections import deque
d = deque((), 2)
```

adritium, 2018-07-26:
That's a max size of 2, not a fixed size.

dhalbert, 2018-07-26:
If you are storing numbers, you could use `array.array`. You could also create a class that implements `__getitem__` and __setitem__` so you could use `[]` subscripts, but there would be no other operations that could change the size

dpgeorge, 2018-07-27:
A `memoryview` has a fixed size.

adritium, 2018-07-27:
@dpgeorge 

Thanks that'll work.

@dhalbert I know I could create a new type but I feel like I've been doing that too much.

closed 2018-07-27 by adritium