micropython/micropython #3983: Unable to connect to ESP8266 via WIFI

hannorein, 2018-07-27:
I am unable to connect reliably to the ESP8266 (the ESP8266 is acting as WIFI access point). The desktop computer claims that the password is incorrect, but I'm using the default `micropythoN` one. Very intermittently, I seem to be able to connect, but I cannot make much sense out of when it works and when not. On the console I see:

```
WebREPL daemon started on ws://192.168.4.1:8266
Started webrepl in normal mode
OSError: [Errno 2] ENOENT

MicroPython v1.9.4-410-g11a38d5dc on 2018-07-27; ESP module with ESP8266
Type "help()" for more information.
>>> add 1
aid 1
station: 54:e4:3a:f0:e8:b9 join, AID = 1
station: 54:e4:3a:f0:e8:b9 leave, AID = 1
rm 1
max connection!
add 1
aid 1
station: f4:0f:24:21:81:39 join, AID = 1
station: f4:0f:24:21:81:39 leave, AID = 1
rm 1
```

I'm using the latest nightly build (older releases show the same issue). I've tried connecting from different computers (all MacOS). Any hint on what might be going on would be appreciated!

peterhinch, 2018-07-28:
I suggest raising this in [the forum](https://forum.micropython.org/) where it may find a larger audience.

dpgeorge, 2018-12-04:
Most likely this is a hardware issue, or issue with wifi range.

closed 2018-12-04 by dpgeorge