micropython/micropython #3984: upip throws error when trying to install umqtt library

Mack13Podwin, 2018-07-27:
I'm using NodeMCU board with 4MB memory and have installed 1.9.4 version of firmware for ESP8266. When i tried to install `umqtt` package via `upip` it has thrown this error:
```error
Error installing 'micropython-umqtt.simple': , packages may be partially installed
```
and file `/lib/umqtt/simple.py` is empty.

dpgeorge, 2018-07-30:
> I'm using NodeMCU board with 4MB memory and have installed 1.9.4 version of firmware for ESP8266. When i tried to install umqtt package via upip it has thrown this error:

Thanks for the report, I can confirm that the ESP8266 will give this error when trying to install umqtt.simple.  When I run the command the error is out-of-memory.  The JSON data for the package is about 10k in size and the ESP8266 doesn't have enough RAM to load and parse the JSON.

> However even with the latest version of upip.py I'm getting a crash on the reference board:

@peterhinch I don't see any hard crash like this.  If you do see such a crash again, can you please dump a bit of the assembler at where it crashed using `xtensa-lx106-elf-objdump -xd build/firmware.elf` and searching for the epc1 hex value (eg search for 4026c578)?

peterhinch, 2018-07-30:
I've rebuilt the firmware since seeing that crash and I can no longer replicate it: upip is working fine.

DevAerial, 2019-04-30:
I'm receiving similar error when running installation on NodeMCU ESP32 with latest `esp32-bluetooth.bin` firmware:
```shell
>>> upip.install('micropython-umqtt.simple')
Installing to: /lib/
Warning: pypi.org SSL certificate is not validated
Installing micropython-umqtt.simple 1.3.4 from https://files.pythonhosted.org/packages/bd/cf/697e3418b2f44222b3e848078b1e33ee76aedca9b6c2430ca1b1aec1ce1d/micropython-umqtt.simple-1.3.4.tar.gz
mbedtls_ssl_handshake error: -71
Error installing 'micropython-umqtt.simple': [Errno 5] EIO, packages may be partially installed
```
Is this error firmware related?

dpgeorge, 2019-05-01:
Error -71 is `MBEDTLS_SSL_ALERT_MSG_INSUFFICIENT_SECURITY`.  Not sure exactly how that is being raised, but I'd suggest just trying again, and trying with the latest firmware.

camp-easy, 2023-10-27:
hello everyone, have you fixed this problem in some way? im using this code with the A9G module:

# Import mqtt (download client if necessary)
try:
    from umqtt import simple
except ImportError:
    import upip
    upip.install("micropython-umqtt.simple")
    from umqtt import simple

and receiving the same error:
Error installing 'micropython-umqtt.simple': [Errno 5], packages may be partially installed

should i use aanother type of code or just reinstalling firmware of A9G? 
Thank you very much in advance...

andrewleech, 2023-10-28:
Hi @camp-easy this issue is _very_ old and things have changed a lot since then. `upip` was replaced a while ago with `mip`
https://docs.micropython.org/en/latest/reference/packages.html

camp-easy, 2023-10-28:
hi @andrewleech thank you for the info, could you please tell me what would the code with mip to replace my one with pip look like?? cause im not understanding well from your link... Thank you in advance for your support.

andrewleech, 2023-10-28:
There's some good advice here: https://www.tomshardware.com/how-to/raspberry-pi-pico-micropython-mip
It should be pretty well compatible with your module if it has working network access.

camp-easy, 2023-10-28:
ok thanks

closed 2023-10-29 by mattytrentini