micropython/micropython #3988: format() on byte string should not be supported

goatchurchprime, 2018-07-29:
In Python3 we have:
 print([b"%02x" % 18])
--> [b'12']

and: 
 print([b"{:02X}".format(18)])
-->  AttributeError: 'bytes' object has no attribute 'format'

however in Micropython:
 print([b"%02x" % 18])
--> [b'12']

and:
 print([b"{:02X}".format(18)])
--> ['12']

This might look like it's just something for language lawyers, but it caught me out as I had converted lots of format strings from % to use the .format() command before I realized it was giving me text strings instead of byte strings.  An error, like in CPython would have been better.  

I don't know why the %-operator was extended to byte strings, but not the .format() command.

pfalcon, 2018-09-08:
I see the issue here is that .format() on `bytes` returns `str` instead of `bytes`, that should be fixed.

Otherwise, CPython indeed has a lot of accidental gaps, MicroPython is more regular, e.g. almost all things which work for `str` also work for `bytes`.

There's also a patch to allow to disable % string formatting https://github.com/micropython/micropython/pull/4061, because having 2 ways to format is indeed too much for small configuration. Then .format() remains the only formatting way, and should work for bytes too.

If you're interested in CPython compatibility, you should test regularly and incrementally using both CPython and MicroPython during development.
Extending .format() to bytes is also something to bring up with them.

dpgeorge, 2019-06-24:
#4181 was merged so the issue here is effectively resolved.

closed 2019-06-24 by dpgeorge