micropython/micropython #3993: docs: updating docs with regard to machine and pyb

rolandvs, 2018-07-30:
Can we expect an update on the documentation of micropython?

Some functions from `pyb` are moved to different libraries like `uos`, `machine`, and `utime` etc. and in some ports the `MICROPY_PY_PYB_LEGACY` flag is used to switch out the calls that were moved out.

If `machine` is still `WIP` should we not keep `MICROPY_PY_PYB_LEGACY` active until the conversion is complete? 

And with the latest doc of the pyboard there is no reference that `pyb.stop()` has become `machine.sleep()` or a cross table, etc.

Docs of `machine.sleep()` is not very informative and "wrong":

`Stops the CPU and disables all peripherals except for WLAN. Execution is resumed from the point where the sleep was requested. For wake up to actually happen, wake sources should be configured first.`

Docs of `pyb.stop()` tells us a lot more: 
` Put the pyboard in a “sleeping” state.
This reduces power consumption to less than 500 uA. To wake from this sleep state requires an external interrupt or a real-time-clock event. Upon waking execution continues where it left off.
See rtc.wakeup() to configure a real-time-clock wakeup event.`

dpgeorge, 2018-07-31:
> Can we expect an update on the documentation of micropython?

Yes it is currently active WIP, see eg #3989.

> If machine is still WIP should we not keep MICROPY_PY_PYB_LEGACY active until the conversion is complete?

Yes it will be enabled until everything is transitioned.  Even after then it may still be kept enabled to help with backwards compatibility.  The idea to introduce it was to make it easier to flag things that are legacy, and allow boards to turn it off to save space if they don't need these legacy functions.  The F091 has it disabled by default to save space, but feel free to turn it on in your own build if you need it.

dpgeorge, 2022-06-23:
The docs have seen a lot of updates and improvements in the past 4 years.  So closing this issue.

closed 2022-06-23 by dpgeorge