micropython/micropython #3994: [pyboard] rtc.wakeup() freezes the system

rolandvs, 2018-07-30:
I tried this on a `pyboard` and it flashes the LED. Same program on the STM32L476_DISC port set the LED and freezes. Adapted it for the NUCLEO_F093RC and it also freezes.

Did anyone tested this functionality or can test this to confirm this behaviour?

```
import pyb
rtc = pyb.RTC()  # rtc is running

def lpdelay(ms):
    rtc.wakeup(ms)
    pyb.stop()
    rtc.wakeup(None)

led = pyb.LED(1)
while True:
    led.on()
    pyb.delay(1000)
    led.off()
    pyb.delay(1)
    lpdelay(1000)
```

For the `NUCLEO_F093RC` port using `machine` instead of `pyb`
```
import pyb, time, machine
rtc=pyb.RTC()

def lpdelay(ms):
    rtc.wakeup(ms)
    machine.sleep()
    rtc.wakeup(None)

led = pyb.LED(1)
while True:
    led.on()
    time.sleep_ms(1000)
    led.off()
    time.sleep_ms(1)
    lpdelay(1000)

dpgeorge, 2018-07-31:
Thanks for the report.  I could confirm that neither `rtc.wakeup()` nor `machine.sleep()` were working for STM32F0 MCUs.  Both of these issues are now fixed by 9dfbb6cc169ab95f07876d97824fd2c3dae229a1 and 21dae87710d8053b2283ef223cb33248bab107b3 respectively.  Now the following script should work the same on pyboard and NUCLEO_F091RC:
```python
import pyb, time, machine
rtc = pyb.RTC()
rtc.init() # needed if RTC is not fully initialised

def lpdelay(ms):
    rtc.wakeup(ms)
    machine.sleep()
    rtc.wakeup(None)

led = pyb.LED(1)
while True:
    led.on()
    time.sleep_ms(1000)
    led.off()
    time.sleep_ms(1)
    lpdelay(1000)
```

As for the STM32L476_DISC, I didn't try that board.  It may have issues with `rtc.wakeup` or `machine.sleep`, or both.

rolandvs, 2018-07-31:
@dpgeorge - I tested it once more with your patches, but still not working on the STM32L476RG.
The sleep state is entered (I see the current drop :-) but no waking up...

rolandvs, 2018-07-31:
@dpgeorge - Also tested the F091 and can confirm that one is working!

rolandvs, 2018-07-31:
@dpgeorge - When wakeup is set through `rtc.wakeup(1000)` the  `EXTI_IMR1` register changes from default `0xff820000` to `0xffc20000`. Bit 22 is being set but should be 20 according to RM0351 page 404. In `extint.h` this seems to be set to the wrong value. But there is probably more to it as changing line 41 `#if defined(STM32F0) || (defined(STM32L4)` did not lead to a working situation :-)

rolandvs, 2018-08-01:
Beautiful! - Battery life in my application was raised from 35 to 637 days.

rolandvs, 2018-08-01:
@dpgeorge - Just a question, I came across this code fragment and was wondering if this could raise problems in the future? While looking into the RTC/EXTI registers I came across RM0351, Page 408 that states that some of the unused bits must keep their reset values (which are unknown). So would be using `EXTI->PR1 |= 1 << EXTI_RTC_WAKEUP` etc. be a better option? 

mp_obj_t pyb_rtc_wakeup(size_t n_args, const mp_obj_t *args) {
        // clear interrupt flags 
        RTC->ISR &= ~RTC_ISR_WUTF; 
        #if defined(STM32L4) 
        EXTI->PR1 = 1 << EXTI_RTC_WAKEUP; 
        #elif defined(STM32H7) 
        EXTI_D1->PR1 = 1 << EXTI_RTC_WAKEUP; 
        #else 
        EXTI->PR = 1 << EXTI_RTC_WAKEUP; 
        #endif 
         NVIC_SetPriority(RTC_WKUP_IRQn, IRQ_PRI_RTC_WKUP);

dpgeorge, 2018-08-01:
> some of the unused bits must keep their reset values (which are unknown). So would be using EXTI->PR1 |= 1 << EXTI_RTC_WAKEUP etc. be a better option?

These registers are "flag" registers and you write a 1 to a certain bit to clear the flag.  So usually for registers like this you don't need to do a read-modify-write.  And indeed in the ST Cube HAL they do a direct write to `PR1`, see eg `LL_EXTI_ClearFlag_0_31`, so I think we are ok doing it.

rolandvs, 2018-08-01:
Using the REPL via the STLINK on the STM32L476DISC board and entering the LED flashing demo from above makes it possible to break the endless loop and get back into the system.

The USB operation is terminated and not restored, so using REPL via a standard UART would be wise.

dpgeorge, 2018-08-02:
The issues with F0 and L4 are now resolved, so closing.

closed 2018-08-02 by dpgeorge