micropython/micropython #3998: Hard Fault in mp_stack_set_top(&_estack);

andrewleech, 2018-08-01:
This is probably caused by something I'm doing wrong, but I'm trying to get micropython head to run on my custom STM32F765 board. I usually run an openmv based fork on this board which works fine.

During `stm32_main`, I'm getting a hard fault when running `mp_stack_set_top(&_estack);`
I've also compiled and tried to run the unmodified boards `NUCLEO_F767ZI` and `STM32F769DISC` which are almost identical chips, and in both cases they also hard fault at the same line.

Does anyone have either a `NUCLEO_F767ZI` or `STM32F769DISC`  board and could check that current head d8e0320  runs on them?

Thanks!

rolandvs, 2018-08-01:
I have one here.

rolandvs, 2018-08-01:
@dpgeorge - Anything that would help your test? 

Observation: when just enabling the LEDs with pyb.LED(1).on() fiddling with the file system from a host will switch the LED off! This is normal I think...

andrewleech, 2018-08-01:
Thanks for checking that, good to know. Something wrong with my build or flashing method then.

The linker script from openmv that works on that fork is just from the OPENMV3 board, however it's suspiciously different to the STM32F767 one in micropython (765 and 767 have same memory map, they share a datasheet). 
I originally built with the openmv one, when it didn't work I switched to the 767 one in micropython which gave the same errors. It's pretty similar to the 769 one also, which I tried to build with... giving the same failure.

I'm just flashing the `firmware.hex` file from build-<BOARD> folder with swd (segger jlink), is that still correct? Or is there a bootloader in master that needs to be flashed as well? I'll try flashing the `firmware.dfu` with the builtin (stm boot rom) usb dfu tomorrow, see if that makes any difference.

Thanks!

andrewleech, 2018-08-02:
Ok, so flashing over DFU has got it working! 
Sorry for the noise.... not sure why flashing over the debugger failed, it's been fine for all my other dev work.

It's not coming up as mass storage, but the usb serial port repl is working fine so the core system is up and running all good. I'm not fussed about mass storage at all, can play with usb config if/when needed.

For reference, this is with a unmodified copy of the 767 linker script and `mpconfigboard.mk`:
```
MCU_SERIES = f7
CMSIS_MCU = STM32F765xx
AF_FILE = boards/stm32f765_af.csv
LD_FILES = boards/stm32f765.ld boards/common_ifs.ld
TEXT0_ADDR = 0x08000000
TEXT1_ADDR = 0x08020000
```
With the `stm32f765_af.csv` also copied from 767 (probably should have just referenced the 767 file instead).

Thanks for the help!

closed 2018-08-02 by andrewleech