micropython/micropython #4001: [discussion] Sharing custom C modules is hard

sdfgeoff, 2018-08-01:
Currently, when writing a custom C module, you have to make edits to the Makefile and mpconfigport.h. Also, the c and h files _must_ exist in the `ports/<yourboardname>` folder - as least so far as I can tell.

This means that, that to distribute a custom C module, you can only do so within the context of a fork of the micropython repository. If I find one person who's written a micropython module for vectormath, and another person who's done a particular communication protocol, then in order to use both of these modules, I have to have a fork of the micropython repository and somehow merge in the code from both of these. There's no easy way to share custom C modules libraries.

I think this contributes hugely to why there are very few custom C modules publicly available for micropython - it's simply too hard to distribute them in a meaningful manner.

Ideally there would be some way of loading compiled libraries (similar to how ctypes allows importing .so files). This is probably impossible in a useful way, but it would be ideal ... allowing drag-drop loading of C modules.
A more realistic way would be to allow subdirectories in the ports folder,  so that at the very least, the c and h files for a module can reside in there. Perhaps offer some way to tie that in to the build system without requiring editing the `mpconfigport.h` and `Makefile`

I'm willing to invest some time into solving this problem, but was wondering if there are other solutions I missed, or if someone has a better idea about how to solve it.

aykevl, 2018-08-01:
There are two (mostly orthogonal) proposals to fix this:

1. #1627, which is a system for dynamically loadable modules (as an extension to .mpy files). The discussion got stuck at what to do with the ESP ports (using the Xtensa architecture) that don't support position-independent code. I also did some work on loading .so files in #3311 but abandoned it in favor of the aforementioned branch by Damien (which is more lightweight).
2. #3871 that I wrote myself, which statically compiles modules in the final binary. See the description for details.  
  This one is more difficult to use than the first one (requiring a recompilation of MicroPython). It is also more powerful as it can call any internal MicroPython function and it is lighter on resources as no dynamic loading (or even a filesystem) is needed for it to work.

Note that both solve roughly the same problem, they do it in very different ways and are targeted at slightly different audiences / types of modules, so I think it's valuable to have both.

sdfgeoff, 2018-08-01:
Hmm, your solution (#3871) looks like it would fit what I was expecting. Pity it hasn't been merged yet. I'll keep my eyes on that PR. Hopefully it gets merged soon.

stinos, 2018-08-02:
> Ideally there would be some way of loading compiled libraries

https://github.com/stinos/micropython/commit/e858818293a0c91f237a93cf63c1306a0e6f575e is similar to the CPython way of loading dynamic libraries, so you give someone a Makefile and a bunch of C (or C++) code which creates a .so which can then be imported as a module. Only tested on unix and windows, not sure how it compares to the other solutions.

adritium, 2018-09-11:
> Anything that's a file on the filesystem needs to be first read into ram before it can be run.

Not true. With a few changes, `mp_raw_code_load` could be made to execute the `mp_raw_code_t` without importing to RAM.