micropython/micropython #4003: [stm32/cc3k] Compilation fails

rolandvs, 2018-08-02:
The current `cc3k` CC3100 support does not compile as it hits an error being type `time_t` be redeclared.

```
In file included from ../../drivers/cc3000/src/patch_prog.c:3:0:
../../drivers/cc3000/inc/cc3000_common.h:168:15: error: conflicting types for 'time_t'
 typedef INT32 time_t;
```

Also there seems to be no support for the CC3200 even though the driver is available? Is this WIP by anyone or did I not look good enough?

dpgeorge, 2018-08-04:
It's strange because it compiles OK on Travis.  Maybe it's due to the version of newlib and system headers.

> Also there seems to be no support for the CC3100 even though the driver is available?

This driver is used by the cc3200 port.  It could be used by other ports and there is some existing work that does this, see eg https://github.com/emfcamp/micropython/blob/tilda-master/stmhal/modnwcc3100.c

Note that the newer version of the CC3100 (the CC3120) uses a completely different driver.

rolandvs, 2018-08-04:
I have updated the PR #4013

dpgeorge, 2018-08-08:
Fixed by c1c798fbc384ff625a1ca3cd0189d0899feb0e4a

closed 2018-08-08 by dpgeorge