micropython/micropython #4008: Threading using _thread causes  segmentation fault, AttributeError:

mzakharocsc, 2018-08-02:
My micropython [program](https://pastebin.com/PTD0qE8U) is using queues and _thread API to schedule processing across multiple cores. It crashes with segfaults, or error messages like these:

> AttributeError: 'Bar' object has no attribute 'lock'
> Segmentation fault (core dumped)

or this:
> ../../py/gc.c:575: gc_free: Assertion `ATB_GET_KIND(block) == AT_HEAD' failed.

or  this:
> ../../py/gc.c:673: gc_realloc: Assertion `ATB_GET_KIND(block) == AT_HEAD' failed.

It sometimes takes hours, and sometimes minutes to run the program and experience a crash. Errors are not consistent, switching between AttributeError, gc_free Assertion, or a quiet segfault. Any suggestions on how to attempt to resolve this would be much appreciated. 

Micropython version: 1.9.4. Target: unix  (8-core x86-64 and 4-core ARM).  My ARM setup seems to reproduce this more often than x86.

aykevl, 2018-08-02:
I can imagine two possible causes:
* The memory allocator (py/gc.c) doesn't do proper locking.
* There is a memory corruption issue somewhere else.

See:
https://github.com/micropython/micropython/blob/master/py/gc.c#L99

dpgeorge, 2018-08-04:
Is the GIL enabled or disabled in your case?  If you are using the standard unix configuration then the answer would be that the GIL is disabled.

jonnor, 2024-09-29:
Information is missing to reproduce/debug the issue. Please provide the requested details, or we will have to close the issue.

closed 2024-10-01 by mzakharocsc