micropython/micropython #4015: ssd1306 display 'rotate' method

robhamerling, 2018-08-03:
For some reason unknown to me the ssd1306.py driver uses the display 'upside down'. Of course this arbitrary, but what I mean: rows and columns are initialised 'mirrored' in __init().
>SET_SEG_REMAP | 0x01, # column addr 127 mapped to SEG0
    SET_COM_OUT_DIR | 0x08, # scan from COM[N] to COM0

I would like to use this display 'upside-up', but the driver has no 'rotate' method (like I found in the sh1106 driver). For example:

    def rotate(self, flag):
        if flag:
            self.write_cmd(SET_COM_OUT_DIR)             # mirror vertically
            self.write_cmd(SET_SEG_REMAP)               # mirror horizontally (64 rows)
        else:
            self.write_cmd(SET_COM_OUT_DIR | 0x08)
            self.write_cmd(SET_SEG_REMAP | 0x01)`

Note: This addition works fine for my 64-rows display, but needs to be improved to support both 32 and 64 rows. May I suggest to add this rotate-facility to the standard driver?

mcauser, 2021-04-20:
Implemented in https://github.com/micropython/micropython/pull/7135

WoodRobotics, 2021-04-28:
Why did I write this? Thanks for saving me some cycles.

```
mirrored_Buffer = bytearray(self.pages * self.width)

for page in range(0, self.pages):
    for column in range(0, self.width):
    
        page_inverse = (self.pages - 1) - page
        column_inverse = (self.width - 1) - column
        
        pixel = self.buffer[(self.width * page) + column]
        pixel_inverse = 0
        
        for index in range(0, 8):
            pixel_inverse = (pixel_inverse << 1) | (pixel >> index & 1)
            
        mirrored_Buffer[(self.width * page_inverse) + column_inverse] = pixel_inverse

```

closed 2022-03-30 by dpgeorge