micropython/micropython #4017: py/objstr.c: mp_obj_str_format_helper() returns structure by value

pfalcon, 2018-08-03:
This is pretty awful (compiler magic behind the scenes, extra stack usage). vstr_t* better be passed as explicit param, allowing explicit allocation control.

dpgeorge, 2018-12-04:
Returning a struct by value allows the compiler to perform more optimisations than it would if it was explicitly passed as a pointer parameter.  My tests showed that stack usage of `mp_obj_str_format_helper()` is less when returning by value, so it can stay as-is.

closed 2018-12-04 by dpgeorge

pfalcon, 2018-12-04:
> Returning a struct by value allows the compiler to perform more optimisations than it would if it was explicitly passed as a pointer parameter.

Perhaps, but overall C compilers are known to cope much better with scalars than with aggregates. So, maybe a particular version of a particular compiler performs more optimizations, but in general, that's not true.

> My tests showed that stack usage of mp_obj_str_format_helper() is less when returning by value, so it can stay as-is.

Stack usage is exactly of concern here, more specifically, the fact that .format() uses more stack (also more heap) than `%`. Which disables things like https://github.com/micropython/micropython-lib/issues/77 .

Maybe naive changing passing by value to passing by pointer increases stack usage, but the purpose of this ticket is to call for optimization in complex for it.