micropython/micropython #4022: [STM32L476] ADCAll not working correctly

rolandvs, 2018-08-04:
Did not look into it in detail yet (source/ref manual), but similar problem as with #3917.
It seems that the STM32L476 reports only the core_vref ~ 1.32V on mine.

```
import pyb
adc=pyb.ADCAll(12, 0x70000)
adc.read_core_vbat()
adc.read_core_temp()
adc.read_core_vref()
adc.read_vref()
```

Could somebody verify this?

rolandvs, 2018-09-19:
@dpgeorge Any change you find some time to check this?

dpgeorge, 2018-09-20:
Thanks for providing the detailed test script.  Looking into it I found two issues:
- The L4 ADC calibration values use 3.0V as a reference, not 3.3V (as do all the other MCUs).  Fixed by 3220cedc31f2ce161286baf58cfdfd396697348a
- The L4 requires a lot longer sample time to get accurate readings from the internal sensors.  With short sample times you get large variations in VREFINT due to cross contamination between channels, which then throws off the other readings, which are corrected against the calibrated VREFINT value.  Fixed by cb3c66e79358f03367d0b4e5d2b6ec649ac29733

@rolandvs see how you go with these fixes.

rolandvs, 2018-09-20:
Great, I 'll put it to the test...

rolandvs, 2018-09-22:
Build fresh version and executed script again did not resolve the problem. HOWEVER this is the effect when using the `rtc.wakeup()` and `pyb.stop()` functions. If these are not used everything is working as expected!!!

I came across this effect, but forgot to report it (so I will do so now): when coming back from a pyb.stop() with rtc.wakeup() it seems that some peripherals are affected (maybe clock disabled?)

```MicroPython v1.9.4-568-g4df194394 on 2018-09-22; L476-DISCO with STM32L476```

USING NO POWER DOWN FUNCTIONS, DISABLED GOSLEEP() IN CODE

(2015, 1, 1, 4, 0, 0, 0, 255)
(2015, 1, 1, 4, 0, 0, 0, 255)
While awake (PC0): 983
ADC core
core Vbat: 3.005582
     Vref: 3.168046
core Vref: 1.215385
core Temp: 15.93354
While awake (PC0): 958
ADC core
core Vbat: 3.003261
     Vref: 3.168046
core Vref: 1.215385
core Temp: 15.93354
While awake (PC0): 956
ADC core
core Vbat: 3.005582
     Vref: 3.168046
core Vref: 1.215385
core Temp: 15.93354

USING THE RTC.WAKEUP() AND PYB.STOP()

From reset:
```
(2015, 1, 1, 4, 0, 0, 5, 227)
(2015, 1, 1, 4, 0, 0, 5, 226)
While awake (PC0): 1066
ADC core
core Vbat: 2.846154
     Vref: 3.172084
core Vref: 1.215385
core Temp: 15.37801
While awake (PC0): 940
ADC core
core Vbat: 2.184439
     Vref: 5.294681
core Vref: 1.215385
core Temp: 209.5621
```

Soft reboot
```
(2015, 1, 1, 4, 0, 0, 5, 241)
(2015, 1, 1, 4, 0, 0, 5, 240)
While awake (PC0): 940
ADC core
core Vbat: 3.646154
     Vref: 5.294681
core Vref: 1.215385
core Temp: 209.5621
While awake (PC0): 940
ADC core
core Vbat: 3.646154
     Vref: 5.294681
core Vref: 1.215385
core Temp: 209.5621
While awake (PC0): 940
ADC core
core Vbat: 3.646154
     Vref: 5.294681
core Vref: 1.215385
core Temp: 209.5621
```

dpgeorge, 2018-09-24:
> when coming back from a pyb.stop() with rtc.wakeup() it seems that some peripherals are affected (maybe clock disabled?)

Indeed, the PLLSAI1 was not reenabled upon waking.  Should be fixed by 1acf58c08f0f29408522a99cc7ef1d6270c21c09

hoihu, 2018-10-31:
It seems that `adc_core.read_core_temp()` is still not working (as indicated in the logs of @robert-hh it outputs something like 217.xx counts). I experience this on my L496 board too.

robert-hh, 2018-10-31:
No me. @rolandvs made the logs.

rolandvs, 2018-10-31:
I've tested it on PYB10, PYB11, F091, F7, H7, L476, but not on L496... as I do not have such board :-(

Using my L476 board:
```
CPU Core Temp: 21.63
CPU Core Vbat: 3.01
CPU Core Vref: 1.21
         Vref: 1.86
```

dpgeorge, 2018-11-01:
@hoihu The L496 looks like it should work just the same as the L476.  To debug the issue can you please run the following on the board that is not working and provide the output:
```python
import time, machine, pyb

print('VREFINT', hex(machine.mem16[0x1fff75aa]))
print('TS_CAL', hex(machine.mem16[0x1fff75a8]), hex(machine.mem16[0x1fff75ca]))

adc = pyb.ADCAll(12, 0x70001)

for i in range(20):
    print(adc.read_channel(0), adc.read_channel(17), adc.read_channel(18))
    time.sleep_ms(100)
```

dpgeorge, 2018-12-04:
> It seems that `adc_core.read_core_temp()` is still not working

I think this would be fixed by ADC calibration done in 9acc32b40f9346f17ed75746b7f9380e4805cef4

closed 2018-12-04 by dpgeorge