micropython/micropython #4024: STM Compile Fails: 'USB_WritePacket'

diginfo, 2018-08-05:
Hi;

Just tried a clean build (first for 2 months) and got the error below, tried this twice always fails at the same point:

```
........
........
CC ../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c
../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c: In function 'USB_WritePacket':
../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c:888:7: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Werror=attributes]
       USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
       ^~~~~~~~~~
../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c: In function 'USB_ReadPacket':
../../lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.c:914:5: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Werror=attributes]
     *(__packed uint32_t *)dest = USBx_DFIFO(0U);
     ^
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:47: build-PYBV11/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.o] Error 1
```

dpgeorge, 2018-08-10:
This should have been fixed by f6f6452b6f336559e4b44c7a49260bd9c2ba684b .  Does the Makefile you have include this fix?  What is the output of `make V=1`.  What compiler are you using?

dpgeorge, 2018-12-04:
I assume this was resolved.

closed 2018-12-04 by dpgeorge