micropython/micropython #4026: http://micropython.org/resources/code-dashboard/ shows zeros since Jan 1, 2018

pfalcon, 2018-08-06:
http://micropython.org/resources/code-dashboard/ shows zeros since Jan 1, 2018

pfalcon, 2018-12-01:
Remains unfixed.

dpgeorge, 2018-12-05:
Thanks for reporting this issue.  The cause was that, after the move of all ports to the ports/ directory, the script that generated the data was not updated and stopped working.  I've now fixed this script and re-run it for all commits since it stopped working.

Unfortunately the plot is not as good as it could be: the compiler has changed a few times over the 4.5 years, along with the hardware that the pystone benchmark runs on, so the data is not truly representative of the evolution of the code.

pfalcon, 2018-12-05:
Great to know it was just a script fix. Surely, it's not suitable for 100% objective comparison over years, but I find it truly useful for both oversee such evolution, even if approximately, and actually track differences over smaller periods of time, like months. Thanks for fixing!

closed 2018-12-05 by pfalcon

dpgeorge, 2018-12-10:
I regenerated the data starting about March 2016, to try and fill in a lot of holes in the plot.  For comparison, the old data is available at http://micropython.org/resources/code-dashboard/index_nov2018.html

In the new data it is noticeable that 1) pystones drops a lot (and is noiser) due to running on different hardware; 2) unix port has a jump up in size due to system defaulting to PIE/PIC.

pfalcon, 2018-12-10:
Thanks for updates.

> unix port has a jump up in size due to system defaulting to PIE/PIC.

I wonder if it might have sense to disable PIE when rebuilding then. Anyway, it's of indicative nature, and not worth spending too much time on it, there're other regressions waiting in queue, e.g. https://github.com/micropython/micropython/issues/4341