micropython/micropython #4027: func.__name__ not implemented

DavidVentura, 2018-08-06:
Hello! Even though [here](https://github.com/micropython/micropython/issues/614) says that this was implemented on 3cc17c6, I do not see this reflected on `MicroPython v1.9.4-8-ga9a3caad0 on 2018-05-11; ESP module with ESP8266`

```
>>> def a():
...     pass

>>> dir(a)
['__class__']

```

is this correct?

stinos, 2018-08-06:
Yes, see the commit message: "Disabled by default.  Enabled on unix and stmhal ports."
So if you need it you'll have to build locally after modifying your mpconfigport.h to have MICROPY_PY_FUNCTION_ATTRS set to (1)

DavidVentura, 2018-08-06:
Whoops. I assumed this was small enough to be enabled by default, given that `repr(fun)` will print the function name anyway. Is that not the case?

stinos, 2018-08-06:
I assume not: looking at the code it adds a member to the structs for bound methods and bytecode functions which will increase size.

DavidVentura, 2018-08-06:
last comment, I promise. What's the point of __class__ on functions? could that not be replaced with something more useful and keep the size?

stinos, 2018-08-06:
No idea what's the pracical use of it but it's internal implementation isn't done by adding an extra field somewhere, instead it's just 2 short lines of code in runtime.c which can also be disabled via MICROPY_CPYTHON_COMPAT if that is somehow needed. So code-size wise it's a non-issue.

closed 2018-08-06 by DavidVentura